debian-pi/raspbian-ua-netinst #427: Installation through wifi (RPi 3)

ThomDietrich, 2016-09-14:
I am using the current state of branch v1.1.x and wanted to incorporate wifi based installation into my system. We followed the instructions given in the PR https://github.com/debian-pi/raspbian-ua-netinst/pull/397 but the file `wpa_supplicant.conf` seems to be ignored.

Is the described method working?
Did somebody test this with the RPi3 wifi module (BCM43438) yet?

Thanks for any help you can provide.

BenjaminHae, 2016-09-14:
It worked for me on the RPi0

Mausy5043, 2016-09-15:
Please check the location of the `wpa_supplicant.conf`. Can you post the result of an `ls`?

VD17593, 2016-09-17:
Did several tries on RPI3 without success.

First of all this question. Can the installation be executed only based on wifi, or does the configuration needs ethernet to configure the wifi?

Tried with `wpa_supplicant.conf ` in `/boot/config` : no success.

Tried also following the procedure described in [bring-your-own-files](https://github.com/debian-pi/raspbian-ua-netinst/tree/v1.1.x#bring-your-own-files) . 
The file `/etc/wpa_supplicant/wpa_supplicant.conf ` is correctly created now.

But the wifi is not working : 
When I do a `sudo ifconfig -a`  , the result I get is : `eth0 `and `lo` , so the wifi module `wlan0` is missing.

diederikdehaas, 2016-09-17:
I haven't tried it yet, but my guess is that it's missing the RPi3's wireless firmware (for starters). I also don't know whether the drivers for that chipset are present.
I have only tried it with external wifi devices and then it *can* work, but there are some requirements.
This is one of the reasons I expect that we will need several alpha/beta releases of the v1.1.x branch before we can release v1.1.0.

VD17593, 2016-09-18:
Got it running !

These are the (minimal) steps needed to get wifi working on RPI3.
(remark : the configuration is made with ethernet connection)

First of all (as described above), the installation is done following the procedure [Bring your own files](https://github.com/debian-pi/raspbian-ua-netinst/tree/v1.1.x#bring-your-own-files) to create the file `/etc/wpa_supplicant/wpa_supplicant.conf`.

So, before first boot, the following file is created on the SD card `config/files/etc/wpa_supplicant/wpa_supplicant.conf
` (with adapted id and pswd) :   

> country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
        ssid="xxx"
        psk="yyy"
        key_mgmt=WPA-PSK
}

After whole configuration, did following steps.

The `dpkg` has to be fixed (don't know why) : 

> sudo dpkg --configure -a
sudo apt-get update

Install the firmware (steps inspired from [minibian](https://minibianpi.wordpress.com/how-to/rpi3/))

> sudo apt-get -y install firmware-brcm80211 pi-bluetooth wpasupplicant

And update the interface : 

> sudo nano /etc/network/interfaces

Add the following text : 

> allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp

Reboot (without ethernet connected).

With `sudo ifconfig -a` the `wlan0` is visible now.
(remark : didn't check bluetooth)

For some reason, for the wifi to work, I had to reboot without the ethernet connected. 
I'm sure an improvement could be found  for this.

Also the content of the file `wpa_supplicant.conf` is the one working on my configuration. It should be tested if other configurations could work with this as well.

ThomDietrich, 2016-10-04:
@diederikdehaas I'll have further testing with the RPi3 I received this weekend. Regarding the alpha/beta statement. I am using the v1.1.x branch as the base of a [preconfigured image provided for a bigger audience](https://github.com/ThomDietrich/openhabian). Do you see a risk in that?

VD17593, 2016-10-04:
What kind of risk do you mean?

Mausy5043, 2017-06-03:
Please try [this release](https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.1.1-beta).
See if it works for you.

Zeerix, 2017-06-03:
I'm using this branch for a while now. It works well on both RPi2 and RPi3 with an external USB WLAN stick (I'm using one by Edimax), but the internal WLAN on RPi3 does not work without additional packages (see #470).

ThomDietrich, 2017-06-03:
Thanks for the heads up @Mausy5043 ! I'm currently occupied with other tasks but happy to see the progress. I'll test when the time is right. Please feel free to close the ticket if you are confident it's resolved.

Mausy5043, 2017-07-25:
There is now also a beta2

Mausy5043, 2017-10-31:
And a beta3 (@ThomDietrich)

Mausy5043, 2017-12-09:
Closing this issue for now, assuming it is resolved. If you feel the closure is in error, please feel free to re-open and add new information.

closed 2017-12-09 by Mausy5043