debian-pi/raspbian-ua-netinst #428: Can't change keyboard layout during installation

D0gi, 2016-09-15:
Hi,

First, it is great that you have developed this software. Thank you, this saves me a lot of work.
Now my issue:
I want to change the keyboardlayout respectively the KEYMAP automatically during installation. Unfortunately I do not get it. I have already read [Issue#226](https://github.com/debian-pi/raspbian-ua-netinst/issues/226), but that is no fix for me,because of the necessary userInput. I try to install the necessary packages over the _installer-config.txt_ with the line:

`packages=nano,keyboard-configuration,kbd,console-data`

and want to set the KEYMAP whit the command `loadkeys de`, after the install.
But unfortunately none of the specified package is installed, not even _nano_. I guess it is due to that during installation appears an interactive user input, right?
That's why I wanted to ask if there is a way to change keyboard layout during installation, respectively if there is an option in _installer-config.txt_ with which you can change the keyboard layout.

Thanks in advance

Mausy5043, 2016-09-15:
The value for packages must be quoted:

`packages="nano,keyboard-configuration,kbd,console-data"`

Mausy5043, 2016-09-15:
Wild guess:
Instead of using `loadkeys` you may want to try this:
 Change the contents of `/rootfs/etc/default/keyboard` from the `post-install.txt`. I think you need to set `XKBLAYOUT` to another value.

 OR

 Copy a working version of `/etc/default/keyboard` into `/config/files/etc/default/keyboard` and modify `/config/files/my-files.txt` appropriately.

D0gi, 2016-09-15:
Thanks for the quick response, I'll try it

diederikdehaas, 2016-09-15:
If it still fails, can you then share your `config.txt` and the installer log file?

D0gi, 2016-09-20:
@Mausy5043 I try your solutions, but unfortunately `(/rootfs)/etc/default/keyboard` did not exist. But even if I copy the file or create it with the right content, the file `keyboard` is not considered. I think this is because keyboard-configuration,kbd and console-data are not installed. Also the quoting: `packages="nano,keyboard-configuration,kbd,console-data"` can't fix it for me.

@diederikdehaas Unfortunately I can only share the `config.txt` because, since 3 days the installation is is always interrupted by a wget-error while the libs are installed, so I can't share a log.

```
[pi1]
kernel=kernel-rpi1_install.img
initramfs installer-rpi1.cpio.gz
device_tree=
[pi2]
kernel=kernel-rpi2_install.img
initramfs installer-rpi2.cpio.gz

```

Mausy5043, 2016-09-20:
First order of business is to get a working installation with all `packages` installed.
The `wget` error bothers me. Perhaps the mirror you are using is not reachable. 
Which version of the installer are you using? 
Which version of Pi are you using?
Have you inspected the SD-card? Just plug it into your PC and check what is on it. Is there no logfile there?

diederikdehaas, 2016-09-20:
Also exact error messages would go a long way, preferably in text, but a picture/photo is fine too.
Are you using `wget` in `post-install.txt` or sth like that and to what url is it trying to `wget` from? If it is a https url, then that is likely the problem.

sidey79, 2016-11-22:
Hi,

I've the same issue as D0gi .. it hangs at packages=... there are serval packages not installed.
console-data is also on the list of missing packages.

If i install it via apt-get it's no problem.
There is not logfile at /boot after instalation, how can i gather more data during the installation?

Mausy5043, 2016-11-23:
Please post the output of `ls -al /boot`

sidey79, 2016-11-24:
```
~$ ls -la /boot
drwxr-xr-x  4 root root   16384 Jan  1  1970 .
drwxr-xr-x 21 root root    4096 Nov 23 23:32 ..
-rwxr-xr-x  1 root root   13065 Nov 23 23:31 bcm2708-rpi-b.dtb
-rwxr-xr-x  1 root root   13328 Nov 23 23:31 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x  1 root root   13101 Nov 23 23:31 bcm2708-rpi-cm.dtb
-rwxr-xr-x  1 root root   14411 Nov 23 23:31 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root   15081 Nov 23 23:31 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   15350 Nov 23 23:22 bcm2710-rpi-cm3.dtb
-rwxr-xr-x  1 root root   17932 Nov 23 23:31 bootcode.bin
-rwxr-xr-x  1 root root      97 Nov 23 23:32 cmdline.txt
-rwxr-xr-x  1 root root  132692 Nov 23 23:28 config-4.4.0-1-rpi2
-rwxr-xr-x  1 root root     173 Nov 23 23:22 config-reinstall.txt
-rwxr-xr-x  1 root root      81 Nov 23 23:32 config.txt
-rwxr-xr-x  1 root root    2509 Nov 23 23:31 fixup_cd.dat
-rwxr-xr-x  1 root root    6480 Nov 23 23:31 fixup.dat
-rwxr-xr-x  1 root root    9713 Nov 23 23:31 fixup_db.dat
-rwxr-xr-x  1 root root    9713 Nov 23 23:31 fixup_x.dat
-rwxr-xr-x  1 root root 5651327 Nov 23 23:30 initrd.img-4.4.0-1-rpi2
-rwxr-xr-x  1 root root     852 Nov 23 23:22 installer-config.txt
-rwxr-xr-x  1 root root 4957251 Nov 23 23:22 installer-rpi1.cpio.gz
-rwxr-xr-x  1 root root 4944812 Nov 23 23:22 installer-rpi2.cpio.gz
-rwxr-xr-x  1 root root 4231408 Nov 23 23:22 kernel7.img
-rwxr-xr-x  1 root root 4128712 Nov 23 23:22 kernel.img
-rwxr-xr-x  1 root root 4316272 Nov 23 23:22 kernel-rpi1_install.img
-rwxr-xr-x  1 root root 4514168 Nov 23 23:22 kernel-rpi2_install.img
-rwxr-xr-x  1 root root    1494 Nov 23 23:31 LICENCE.broadcom
drwxr-xr-x  2 root root    8192 Nov 23 23:31 overlays
-rwxr-xr-x  1 root root     459 Nov 23 23:22 post-install.txt
-rwxr-xr-x  1 root root  615896 Nov 23 23:31 start_cd.elf
-rwxr-xr-x  1 root root 4934920 Nov 23 23:31 start_db.elf
-rwxr-xr-x  1 root root 2743224 Nov 23 23:31 start.elf
-rwxr-xr-x  1 root root 3887400 Nov 23 23:31 start_x.elf
-rwxr-xr-x  1 root root    3857 Nov 23 23:22 system-config.sh
-rwxr-xr-x  1 root root    5504 Nov 23 23:22 system-install.sh
-rwxr-xr-x  1 root root 2177376 Nov 23 23:28 System.map-4.4.0-1-rpi2
drwxr-xr-x  2 root root    2048 Nov 23 23:22 System Volume Information
-rwxr-xr-x  1 root root 4514168 Nov 23 23:28 vmlinuz-4.4.0-1-rpi2
```

Mausy5043, 2016-11-25:
This:
```
$ dpkg -l keyboard-configuration
dpkg-query: no packages found matching keyboard-configuration
$ sudo apt list |grep keyboard-configuration
keyboard-configuration/stable 1.123 all
```
suggests you can use `chroot /rootfs /usr/bin/apt-get -y install keyboard-configuration` in the `post-install.txt`

Would that help?

aries26, 2017-02-17:
Add my vote for a keyboard-configuration parameter in the installer-config file.

I am preparing a new install where I will have to keep the default kb which is not my preference.

CheshireCat44, 2017-03-25:
Sorry I didn't see this comment earlier. I append an install-config.txt file and a post-install.txt that create the keyboard I want. If it works for someone else, great!
The post-install.txt file contains sections extracted from the rcS file in the 1.1.x branch

Many thanks to those who have developed the unattended install system.

Cheshire Cat 

[installer-config.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/870402/installer-config.txt)
[post-install.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/870403/post-install.txt)

Mausy5043, 2017-03-26:
This looks really nice. But, note that the locale installation is present in the `v1.1.x` tree. So, there's no need to copy _that_ code if you build your installer based on the `v1.1.x`-tree.

CheshireCat44, 2017-03-26:
Given that I had copied the code from the 1.1.x tree originally, I was aware of that some months ago but had forgotten it. Thanks for the reminder. The trouble is that the 1.0.x tree works for me at present so there is less incentive to change. "The good is the enemy of the best."

Mausy5043, 2017-03-26:
@CheshireCat44 thanks for your input. Your `post-install.txt` file provided above can hopefully serve as a sample for others to modify their own installation in case they need a different keyboard than the one provided by default (US).

Mausy5043, 2017-03-26:
Closing this issue. This is a minimal installer that provides a default locale and keyboard configuration. The user may post-install a different keyboard if the need arises using the sample code provided in this issue.
If you feel the closure was in error. Please feel free to re-open and add your comment/info.

closed 2017-03-26 by Mausy5043