debian-pi/raspbian-ua-netinst #430: Add files to be put onto the installation?

Jodes81, 2016-09-23:
I want to add files to the installer which would be copied onto the final installation. They might include:

Configuration files that replace those in the default installation once installation is complete. (Which I would trigger using _post‑install.txt_.) E.g. /root/.bashrc, samba config files, etc.

Other files might be PHP scripts, Node.JS files, images, css files, compiled C binaries or what have you. 

Is this already possible? If not, how could it be done?

Jodes81, 2016-09-23:
All files loaded onto the SD card prior to installation are copied into /boot/

closed 2016-09-23 by Jodes81

Mausy5043, 2016-09-24:
As of v1.1.x you can put your files in `config/files`. Look for "Bring your own files" in the `README.me` of v1.1.x.

Mausy5043, 2016-09-24:
@Jodes81 ^^