debian-pi/raspbian-ua-netinst #432: no space left on device (16gb microsd)

aristosv, 2016-09-28:
**This is how my installation starts:**

```
=================================================
raspbian-ua-netinst
=================================================
Revision git~6171a93
Built on Wed 28 Sep 20:00:49 EEST 2016
Running on Raspberry Pi version 3
=================================================
https://github.com/debian-pi/raspbian-ua-netinst/
=================================================
Starting HWRNG succeeded!
Copying boot files... OK
=================================================
=== Start executing installer-config.txt. ===
=== Finished executing installer-config.txt. ===
=================================================

Network configuration:
  ip_addr = dhcp
  online_config = 

Waiting for eth0... 
Configuring eth0 with DHCP... 192.168.7.222
Time set using ntpdate to... Wed Sep 28 18:14:29 UTC 2016

Installation started at Wed Sep 28 18:14:14 UTC 2016.

Installer configuration:
  preset = base
  packages = ifupdown,net-tools,openssh-server,iputils-ping,wget,ca-certificates,cron,ntpdate,nano,lftp,mpg123,sshpass,bc,psmisc,alsa-utils,curl,mailutils
  mirror = http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian
  release = jessie
  hostname = ClientX
  domainname = 
  rootpw = raspbian
  root_ssh_pubkey = 
  disable_root = 
  username = 
  userpw = 
  user_ssh_pubkey = 
  user_is_admin = 
  cdebootstrap_cmdline = --flavour=minimal --include=linux-image-rpi2-rpfv,cpufrequtils,kmod,libpam-systemd,rng-tools,isc-dhcp-client,ifupdown,net-tools,openssh-server,iputils-ping,wget,ca-certificates,cron,ntpdate,nano,lftp,mpg123,sshpass,bc,psmisc,alsa-utils,curl,mailutils
  boot_volume_label = 
  bootsize = +64M
  bootoffset = 8192
  rootsize = 
  timeserver = time.nist.gov
  timezone = Etc/UTC
  locales = 
  system_default_locale = 
  cmdline = dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline
  usbroot = 
  rootdev = /dev/mmcblk0
  rootpartition = /dev/mmcblk0p2
  rootfstype = ext4
  rootfs_mkfs_options = 
  rootfs_install_mount_options = noatime,data=writeback,nobarrier,noinit_itable
  rootfs_mount_options = errors=remount-ro,noatime
  final_action = reboot
  hardware_versions = 3
  gpu_mem = 
```

**and this is how it ends**

```
ln: 
failed to create hard link '/boot/initrd.img-4.4.0-1-rpi2.dpkg-bak' => '/boot/initrd.img-4.4.0-1-rpi2'
: Operation not permitted

gzip: stdout: No space left on device
update-initramfs: failed for /boot/initrd.img-4.4.0-1-rpi2 with 1.
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
E: Internal error: install

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

Which is strange, saying that there's no space left. Its a 16GB microsd. On the rpi2 I usually run the script on a 4GB microsd and its more than enough.

This is the whole log file. http://paste.ofcode.org/eQdaDhgxPe2XxVVCLjFdTp
Any idea what could I be doing wrong?

kpfleming, 2016-09-28:
The default for 'bootsize' is too small, I think. I usually set it to +256M, to leave room for future kernel updates after installation.

diederikdehaas, 2016-09-28:
> `bootsize = +64M`

The default is now 128M and we've increased that for a reason :wink: 
Furthermore if you use lots/big custom config/install/etc files to use in `post-install.txt` for example, those get copied to `/boot/` as well and thus take up space.

aristosv, 2016-09-29:
This was it. Thanks

closed 2016-09-29 by aristosv