debian-pi/raspbian-ua-netinst #433: audio problem - module error - file not found

aristosv, 2016-10-03:
On the Raspberry Pi 2 and on the v1.0.x branch, I installed psmisc & alsa-utils, and I was able to play music. For the past few days I've been trying the v1.1.x branch on the Raspberry Pi 3, and when trying to play music using mpg123, I get the following messages.

```
user@Client003:/home/user/scripts# /usr/bin/mpg123 -Z -q -f 57236 /home/user/media/*
[module.c:129] error: Failed to open module jack: file not found
[module.c:129] error: Failed to open module portaudio: file not found
[module.c:129] error: Failed to open module pulse: file not found
[module.c:129] error: Failed to open module nas: file not found
[module.c:129] error: Failed to open module openal: file not found
[audio.c:180] error: Unable to find a working output module in this list: alsa,oss,jack,portaudio,pulse,nas,openal
[audio.c:545] error: Failed to open audio output module
[mpg123.c:913] error: Failed to initialize output, goodbye.
```

I'm guessing it's something I didn't install? Or compile?

aristosv, 2016-10-03:
According to [this](http://forums.debian.net/viewtopic.php?t=49132#p283148) the problem is that mpg123 is not looking at the correct path for the libraries it needs.

So theoretically this should fix the issue (haven't tried it yet)
`export LD_LIBRARY_PATH=/usr/lib:/usr/lib/mpg123`

aristosv, 2016-10-04:
Thomas from mpg123 tells me that my problem is that the soundcard is not installed, and that I might have to install a driver for it. Do you guys have any idea what must I install to get the soundcard to work? Obviously psmisc & alsa-utils are not doing it on the RPi3 as they did it on the RPi2.

diederikdehaas, 2016-10-04:
Search for an overlay you need to specify to enable audio. The Raspberry Pi forums seems a likely candidate, but an internet search should do it too.

aristosv, 2016-10-05:
just an fyi, omxplayer seems to be playing just fine, through HDMI and 3.5 audio jack. I notified mpg123 developer, I'll update this accordingly.

aristosv, 2016-10-19:
No one knows what the problem is. I just gave up on this and now I'm using omxplayer to play mp3 and mp4 files. Its even better now because my scripts are more consistent and I can use the same script on audio players and video players.

closed 2016-10-19 by aristosv

magdesign, 2016-11-15:
Had the same problem. Since they changed some stuff in the kernel, you must add to:

**/boot/config.txt**

a line containing:

**dtparam=audio=on**

After a reboot it will work.