debian-pi/raspbian-ua-netinst #434: dpkg: warning: ignoring pre-dependency problem

ThomDietrich, 2016-10-04:
Hey, working with v1.1.x I'm getting a lot of these warnings:
```
dpkg: warning: ignoring pre-dependency problem!
```
Have a look: http://pastebin.com/HNbVxjGr

I am not sure how to rate them. Is there a problem or are these resolved later? Is there a solution or should these warnings be omitted in the latter case?

wannadrunk, 2016-10-04:
Hi, I couldn't remember which issue# that said we can ignore those warning.

wannadrunk, 2016-10-04:
Got it! It said in README under Installing Note

> Note: During the installation you'll see various warning messages, like "Warning: cannot read table of mounted file systems" and "dpkg: warning: ignoring pre-dependency problem!". Those are expected and harmless.

ThomDietrich, 2016-10-04:
Oh, shame on me for not doing a search first. Thanks

closed 2016-10-04 by ThomDietrich