debian-pi/raspbian-ua-netinst #435: installing raspberrypi-kernel

silberzwiebel, 2016-10-28:
Hi,

I tried to install a recent kernel that includes a fix for the security issue known as dirty cow. Apparently, the package `raspberrypi-kernel` already contains the fix (see https://github.com/raspberrypi/linux/issues/1694)
I installed raspbian with your ua-netinst some time ago and this gave me 
```
Linux pi 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi19 (2015-08-08) armv6l GNU/Linux
```

Although this kernel seems somehow to differ from the raspberrypi-kernel package, I thought installing the raspberrypi-kernel package would give me the correct kernel with the bugfix.
So, I did a 
```
sudo apt-get update
sudo apt-get install raspberrypi-kernel
```

Installation ran fine, no error messages. Then, I rebooted my pi (1, btw), to never get it back online again. After that I messed around with /boot/config.txt, /boot/cmdline.txt, and several other files that I guess I wasn't supposed to touch. At the moment, I'm restoring a backup of my SD card to finally have a running system again (I accidentally triggered a raspbian-ua-netinst re-install and lost everything on the card....)

So - now that I do have a running pi again - what is the proper way to install the `raspberrypi-kernel` package?

Mausy5043, 2016-10-28:
You must *NOT* install `raspberrypi-kernel`. That fix is only valid for Raspberry Pi Foundation installs.

To install a newer kernel you might consider using the procedure below (your warranty is void from this point onwards :wink: ):
```
sudo apt update
sudo apt install rpi-update
sudo rpi-update 230f26174839d79cc706dd5a809611f25967e554
sudo sed -i 's/kernel=/#kernel=/' /boot/config.txt
sudo sed -i 's/initramfs/#initramfs/' /boot/config.txt
```
Then reboot your Pi.

This installs the kernel 4.4.27+ which does not have the [Dirty COW](https://dirtycow.ninja/) vulnerability.

The long string of numbers and characters refers to a commit on [Hexxeh's repository](https://github.com/Hexxeh/rpi-firmware/commits). If you want to update your kernel with this procedure you are advised to only use the commits starting with "kernel: Bump" as those are considered stable enough. The others are more bleeding-edge and may break your system. Click on the clipboard icon on the right to copy the commit-ID to your paste-buffer.

silberzwiebel, 2016-10-28:
Okay, thanks. If I now got it right, the kernel running in a `raspbian-ua-netinst` install comes from the `linux-image-rpi-rpfv` package, correct? How long would it take that an updated kernel arrives in that package so that I can simply run
```
sudo apt-get update
sudo apt-get upgrade
sudo shutdown -r now
```
to be on the safe side?

Mausy5043, 2016-10-28:
> How long would it take that an updated kernel arrives in that package so that I can simply run ...[code]...

Infinity. `sudo apt-get upgrade` does not upgrade the kernel :wink:

silberzwiebel, 2016-10-28:
So the easiest way for me to be safe from dirty COW without reinstalling everything is to use `rpi-update` as you described?

Mausy5043, 2016-10-28:
For now, yes. The installer tends to provide an installed system with a fixed kernel version. If you want a newer kernel `rpi-update` is *a* way forward. There may be others, that I'm not familiar with.

However, be aware that the vulnerability has been there some time and is only really an urgent problem if your Pi is directly facing the internet. If your Pi has a local IP and it has no ports forwarded through the router you needn't worry... much.

silberzwiebel, 2016-10-28:
My pi is facing the internet and ports are forwarded to it through the router, this is why I want to fix it as soon as possible. I'll try the rpi-update now. Thanks for the help!

diederikdehaas, 2016-10-28:
> sudo apt-get upgrade does not upgrade the kernel

That's actually not correct. But plugwash has to make a new kernel package and his availability is "somewhat" limited.

@silberzwiebel are you still running `wheezy`? You really should upgrade to `jessie` and that would at least give you a 4.4 kernel, but (afaik) not with the DirtyCOW fix

silberzwiebel, 2016-10-28:
I'm on jessie (afaik) and running the kernel mentioned in my first post, 3.18.0.

Mausy5043, 2016-10-28:
@diederik : :ok:  I thought only `apt-get dist-upgrade` would up the kernel

Mausy5043, 2016-10-28:
@silberzwiebel Consider building from the v1.1.x installer

diederikdehaas, 2016-10-28:
@Mausy5043 nope.

@silberzwiebel can you post the output of `aptitude search ~ilinux-image`?

silberzwiebel, 2016-10-28:
> @silberzwiebel Consider building from the v1.1.x installer

How would I do that? Does it preserve the contents of my SD card?

> @silberzwiebel can you post the output of aptitude search ~ilinux-image?

```
$ aptitude search ~ilinux-image
i   linux-image-3.18.0-trunk-rpi                                       - Linux 3.18 for RaspberryPI                                                   
i A linux-image-4.4.0-1-rpi                                            - Linux 4.4 for RaspberryPI                                                    
i   linux-image-rpi-rpfv                                               - This metapackage will pull in the raspbian kernel for the raspberry pi 1 
```

diederikdehaas, 2016-10-28:
That was indeed the other possibility :stuck_out_tongue: 

You can and probably should switch to the 4.4 kernel by modifying your `/boot/config.txt` like this:
```
diederik@pi-1bplus:~$ cat /boot/config.txt 
[pi1]
kernel=vmlinuz-4.4.0-1-rpi
initramfs initrd.img-4.4.0-1-rpi followkernel
```
That will give you the latest kernel from raspbian.org. But that does NOT fix the DirtyCOW issue!
I'll try to see whether I can successfully install/run the `raspberrypi-kernel`

Mausy5043, 2016-10-28:
>> @silberzwiebel Consider building from the v1.1.x installer

> How would I do that? Does it preserve the contents of my SD card?

Contents of the SD-card would be destroyed. Let's first see if @diederikdehaas  comes up with an easier solution.

diederikdehaas, 2016-10-28:
I know why you got a non-functional system after trying to install `raspberrypi-kernel` and once again `apt-get` turns out not to be so smart :-/
```
root@pi-1bplus:/home/diederik# aptitude install raspberrypi-kernel
The following NEW packages will be installed:
  raspberrypi-kernel 
0 packages upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 30,0 MB of archives. After unpacking 107 MB will be used.
The following packages have unmet dependencies:
 raspberrypi-bootloader-nokernel : Conflicts: raspberrypi-kernel but 1.20161020-1 is to be installed.
The following actions will resolve these dependencies:

     Remove the following packages:   
1)     libraspberrypi-bin             
2)     libraspberrypi0                
3)     raspberrypi-bootloader-nokernel

Accept this solution? [Y/n/q/?] n
The following actions will resolve these dependencies:

     Keep the following packages at their current version:
1)     raspberrypi-kernel [Not Installed]                 

Accept this solution? [Y/n/q/?] n
The following actions will resolve these dependencies:

     Remove the following packages:                                                      
1)     raspberrypi-bootloader-nokernel                                                   

     Install the following packages:                                                     
2)     raspberrypi-bootloader [1.20161020-1 (stable)]                                    

     Upgrade the following packages:                                                     
3)     libraspberrypi-bin [1.20160523-1~nokernel2 (now, stable) -> 1.20161020-1 (stable)]
4)     libraspberrypi0 [1.20160523-1~nokernel2 (now, stable) -> 1.20161020-1 (stable)]
```

It is extremely likely that `apt-get` chose the first option, thus removing the firmware files which results in a non-bootable system.
I'm now going to say 'y' to that last option and reboot after that (after modifying `/boot/config.txt` so it actually uses the `raspberrypi-kernel` kernel)

silberzwiebel, 2016-10-28:
> It is extremely likely that apt-get chose the first option, thus removing the firmware files which results in a non-bootable system.

Indeed, I ended up with no firmware files. Due to my little knowledge, I copied them from https://github.com/raspberrypi/firmware to my SD card. Since I also (more or less) randomly changed the content of `/boot/config.txt` I ended up with a booting system (hooray!) that re-installed everything (oh, no, stop it, I do need the files on the SD card!). That's the price of being naíve and forgetting what system I'm actually running.

diederikdehaas, 2016-10-28:
And this does indeed work.
So what you should do is the following as **root** (or prepend with `sudo`):
```
# apt-get install raspberrypi-kernel raspberrypi-bootloader
# sed -i 's/kernel=/#kernel=/' /boot/config.txt
# sed -i 's/initramfs/#initramfs/' /boot/config.txt
```
The first command should **un**-install `raspberrypi-bootloader-nokernel` and install the kernel + firmware from the RaspberryPi Foundation.
The last 2 `sed` statements disable booting with the raspbian.org kernel and thus uses the RPF kernel.

@Mausy5043 could you edit your first response to point to my previous reply which shows why it failed for you? I **REALLY** do not like using (or promoting in any way) `rpi-update` at all, even though I understand why you suggested it and it should indeed work.

diederikdehaas, 2016-10-28:
> Indeed, I ended up with no firmware files. Due to my little knowledge, I copied them from https://github.com/raspberrypi/firmware to my SD card.

It's a (little) bit risky but mostly a lot of work, but it should/would indeed work.

silberzwiebel, 2016-10-28:
Thanks for the elaborated answers!

Apart from being safe from the dirty COW issue, are there any pros / cons for using the kernel from the RaspberryPi foundations instead of the kernel from raspbian.org?
Put differently, if the fix for the raspbian.org kernel is released: should I switch back to that kernel? Or should I wait for the release?

diederikdehaas, 2016-10-28:
> are there any pros / cons for using the kernel from the RaspberryPi foundations instead of the kernel from raspbian.org?

Yes.
Pro:

- The Raspberry Pi Foundation updates their kernel more frequently, so you would get fixes earlier

Con:

- The Raspberry Pi Foundation updates their kernel more frequently, which is not what what a Debian/raspbian **stable** release would do. Debian wouldn't upgrade the stable kernel to a new 'major' version, period. Raspbian has compromised on this a bit by upgrading to the 4.4 kernel, but that was needed to support the RPi 3.
Newer versions can be good/fun, but they also introduce new bugs and/or *just* change functionality, which can make your appliance stop working all of a sudden. See http://www.catb.org/esr/faqs/stratum-1-microserver-howto/ for an example.

I am a **big** fan of Debian and therefor prefer Debian's policy of updating software, which raspbian.org sticks to as much as possible.
Some people don't like it and an often heard complaint about Debian stable is that it (over time) contains 'outdated' software. That may be true, but it also means that it is rock solid. But I also think that the 'outdated' software argument is overblown as most people don't need to run the very latest versions to use it.
And if you want to run more recent software, Debian (and raspbian.org) provide a way for that, namely using `testing`/`stretch` or on Debian only `unstable`/`sid`. But you have to accept and capable of dealing with the occasional breakage.

So, as usual, it depends. For seq the kernel, it's probably not much of a problem to use the one from the RPF *for most people*

silberzwiebel, 2016-10-29:
I've succesfully updated my kernel to
```
Linux pi 4.4.26+ #915 Thu Oct 20 17:02:14 BST 2016 armv6l GNU/Linux
```

I was a bit skeptical because the apt-get command also uninstalled the packages libraspberrypi0 and libraspberrypi-bin and I wasn't sure, if they were needed. Apparently, they are not needed (at least for my use case and the boot process in tiself).
Thanks again for your help!

closed 2016-10-29 by silberzwiebel

diederikdehaas, 2016-10-29:
> uninstalled the packages libraspberrypi0 and libraspberrypi-bin

You may want to install those again as they contain the various libraries and programs specifically for the RPi (esp. the GPU).

kpfleming, 2016-10-29:
It's not possible to install those packages, as they depend on raspberrypi-bootloader-nokernel.

diederikdehaas, 2016-10-29:
Try it like this:
```
aptitude install libraspberrypi0=1.20161020-1 libraspberrypi-bin=1.20161020-1
```
Then it will install the versions from the RPF and those depend on raspberrypi-bootloader

Mausy5043, 2016-10-29:
To sum it up. This is what seems to work (for me at least):

```
sudo apt-get update
sudo apt-get -yuV remove raspberrypi-bootloader-nokernel
# clean up 
sudo apt-get -y autoremove
sudo apt-get -y autoclean
sudo apt-get -yuV install raspberrypi-bootloader
sudo apt-get -yuV install libraspberrypi0=1.20161020-1 libraspberrypi-bin=1.20161020-1
# let's check what we got
apt list --installed |grep raspberry
```

which results in these installed packages:
```
libraspberrypi-bin/stable,now 1.20161020-1 armhf [installed]
libraspberrypi0/stable,now 1.20161020-1 armhf [installed]
raspberrypi-bootloader/stable,now 1.20161020-1 armhf [installed]
raspberrypi-kernel/stable,now 1.20161020-1 armhf [installed,automatic]
```

And after reboot:
```
$ uname -rv
4.4.26+ #915 Thu Oct 20 17:02:14 BST 2016
```

I do, however, dislike having to explicitly set the version of both `libraspberry*` packages.

Mausy5043, 2016-10-29:
Could this be the cause?

```
$ cat /etc/apt/preferences.d/01vcgencmd.pref
Package: libraspberrypi-bin libraspberrypi0 libraspberrypi-dev libraspberrypi-doc
Pin: release o=Raspbian,c=firmware
Pin-Priority: 800
```

diederikdehaas, 2016-10-29:
That is indeed very likely it.
Without it, it would take the latest version, which is the one from RPF

diederikdehaas, 2016-10-29:
Do an `apt-cache policy libraspberrypi-bin libraspberrypi0` before and after removing/disabling that file to make sure

kpfleming, 2016-10-30:
Confirmed on my systems; removing that preferences file results in the RPF versions of these packages being installed.