debian-pi/raspbian-ua-netinst #436: Kernel updates

typhoon71, 2016-11-02:
I don't really know if this is an issue with the netinst, but I can't seem to update (upgrade) the kernel.
I did create a working SD some time ago with weezy on it, ans wanted to upgrade the kernel.
I noticed there aren't any package in the repos, but since I was able to update it with rpi-update on another SD I made I tried with it; it seemed to properly download it but then it doesn't use it.
Any idea?

diederikdehaas, 2016-11-02:
It could be that wheezy didn't get the new kernel and I'd urge you to upgrade to jessie in any case.
Jessie does have a 4.4 kernel, but even if the kernel package is upgraded, it won't be loaded by default as you need to adapt `/boot/config.txt` to point to the 4.4 equivalents of the entries you see there.

Also take a look at https://github.com/debian-pi/raspbian-ua-netinst/issues/435 which addresses a similar issue.

typhoon71, 2016-11-02:
Thanks for the info, I didn't know about the ditry cow thing.
Sadly I won't be able to access the rpi soon, so I can't reinstall the whole thing upgrading to jessie. :P

diederikdehaas, 2016-11-02:
By changing `/etc/apt/sources.list` by replacing `wheezy` with `jessie` and doing a `apt-get dist-upgrade` or `aptitude full-upgrade` you should be able to **upgrade** to jessie.
But access to your RPi would be needed :stuck_out_tongue: 

I'm closing this issue. If you still have this problem after having performed the mentioned procedure(s), feel free to reopen it again.

closed 2016-11-02 by diederikdehaas

typhoon71, 2016-11-02:
I will do a dist upgrade then, better than leaving the rpi vulnerable.
And it's unsuccessfull I'll have someone to turn it off.
Thanks again for the tip.