debian-pi/raspbian-ua-netinst #438: Pi Zero, LAN Connection

BenjaminHae, 2016-11-18:
I bought this [3-port hub with ethernet](https://shop.pimoroni.com/products/three-port-usb-hub-with-ethernet-and-microb-connector). Starting the installer leads to "waiting for eth0", but it is connected. After the countdown and the waiting for input to open a shell the installation finishes.

I'd gladly give more information when you tell me what to find out ;)
A log file has not been created as far as I can see. My `installer-config` just contains
```
packages=unionfs-fuse,curl,unzip,screen
```

Mausy5043, 2016-11-18:
Can you show a listing of the SD contents?

BenjaminHae, 2016-11-18:
Oh I forgot(but the pi zero implies): I'm on branch v1.1.x.
```
insgesamt 43358
-rw-r--r-- 1 user user     2354 Aug 23 10:47 authorized_keys
-rw-r--r-- 1 user user    13065 Nov 18 18:41 bcm2708-rpi-b.dtb
-rw-r--r-- 1 user user    13328 Nov 18 18:41 bcm2708-rpi-b-plus.dtb
-rw-r--r-- 1 user user    13101 Nov 18 18:41 bcm2708-rpi-cm.dtb
-rw-r--r-- 1 user user    14411 Nov 18 18:41 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 user user    15081 Nov 18 18:41 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 user user    17932 Nov 18 18:41 bootcode.bin
-rw-r--r-- 1 user user       73 Nov 18 18:41 cmdline.txt
drwxr-xr-x 3 user user     2048 Nov 18 18:41 config
-rw-r--r-- 1 user user      294 Nov 18 18:41 config.txt
-rw-r--r-- 1 user user     2509 Nov 18 18:41 fixup_cd.dat
-rw-r--r-- 1 user user     6480 Nov 18 18:41 fixup.dat
-rw-r--r-- 1 user user     9713 Nov 18 18:41 fixup_db.dat
-rw-r--r-- 1 user user     9713 Nov 18 18:41 fixup_x.dat
-rw-r--r-- 1 user user       40 Nov 18 18:35 installer-config.txt
-rw-r--r-- 1 user user 11603248 Nov 18 18:41 installer-rpi1.cpio.gz
-rw-r--r-- 1 user user 11612301 Nov 18 18:41 installer-rpi2.cpio.gz
-rw-r--r-- 1 user user  4316272 Nov 18 18:41 kernel-rpi1_install.img
-rw-r--r-- 1 user user  4514168 Nov 18 18:41 kernel-rpi2_install.img
-rw-r--r-- 1 user user     1494 Nov 18 18:41 LICENCE.broadcom
-rw-r--r-- 1 user user     1081 Nov 17 18:57 LICENSE
-rw-r--r-- 1 user user      181 Nov 17 18:57 mount_unionfs
drwxr-xr-x 2 user user     8192 Nov 18 18:41 overlays
-rw-r--r-- 1 user user      171 Nov 17 19:13 poisontap.service
-rw-r--r-- 1 user user     1469 Nov 18 18:49 post-install.txt
-rw-r--r-- 1 user user      819 Jan  1  1980 raspbian-ua-netinst-19700101T000021.log
-rw-r--r-- 1 user user     1317 Nov 17 18:57 README.md
-rw-r--r-- 1 user user   615896 Nov 18 18:41 start_cd.elf
-rw-r--r-- 1 user user  4934920 Nov 18 18:41 start_db.elf
-rw-r--r-- 1 user user  2743224 Nov 18 18:41 start.elf
-rw-r--r-- 1 user user  3887400 Nov 18 18:41 start_x.elf
```
I'm using the read only installation: https://github.com/goranche/raspbian-ua-netinst-ro
But that shouldn't matter as it only gets executed in post-install.

BenjaminHae, 2016-11-18:
And I feel stupid for not seeing the log until after posting:
```
=================================================
raspbian-ua-netinst
=================================================
Revision git~08121b4
Built on Di 23. Aug 10:00:58 CEST 2016
Running on Raspberry Pi version 0
=================================================
https://github.com/debian-pi/raspbian-ua-netinst/
=================================================
Starting HWRNG succeeded!
Copying boot files... OK
=================================================
=== Start executing installer-config.txt. ===
=== Finished executing installer-config.txt. ===
=================================================

Network configuration:
  ip_addr = dhcp
  online_config = 

Waiting for eth0... 
1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. FAILED

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

Mausy5043, 2016-11-18:
Apparently `eth0` is not discovered by the Pi.

Let's first exclude a hardware problem. (ethernet cable, the hub and the Pi's USB port)
Please confirm that both the ethernet cable AND the hub are not defective:
Try plugging the adapter **with** ethernetcable in another device/computer.
Please confirm that the Pi has a working USB-port (`lsusb`)

BenjaminHae, 2016-11-19:
I couldn't try the hub with another computer as it has a micro usb connection.

The ethernet cable for itself does work.
A keyboard connected to the hub does work so I assume the pi's USB-port does work. Running lsusb in the installer outputs `lsusb: not found`. The only thing I could not check is the ethernet port. Which seems to be the most important thing.

I am going to install noobs. Maybe from there I can find out more.

kpfleming, 2016-11-19:
It is very likely that the default kernel does not include the driver for the USB-Ethernet chip inside this device, but it is in a module. You'd need to find out the module name and tell the installer to load that module via the proper setting in the config file.

BenjaminHae, 2016-11-19:
@kpfleming yes that's it, thanks.
For anyone else with similar questions:
Using the raspbian image I used `lsusb -t` to get the necessary driver. It's `r8152`. I specified the driver in `installer-config.txt` in the line `drivers_to_load=r8152`.

BenjaminHae, 2016-11-19:
Alright that didn't work. There doesn't appear an error about the driver, but eth0 is still not found. Do I need to copy the driver to my sd card before? How would I do that?

kpfleming, 2016-11-19:
You can try copying the module into 'lib/modules' in the SD card, ensuring that it's in the same subdirectory path as it appears in the normal /lib/modules directory on a regular installation. Alternatively, you can edit build your own installer after adding another INSTALL_MODULES line in ```build.sh``` to have the build process copy the module into the necessary location.

BenjaminHae, 2016-11-19:
Alright I built it containing `r8152`. Still the same error. I just looked in `/lib/modules/4.4.0-1-rpi/kernel/drivers/net/usb/` and the module is really there.
But `lsmod` show an empty list. After running `modprobe r8152.ko` the list contains the module.
What do I do next?

BenjaminHae, 2016-11-19:
I just forgot to add "drivers_to_load=r8152" again. Thanks for all your help
Now the final way for anyone else with a similar error:

1. Find out what module is needed(easiest way to do this is using a normal rasbpian(-lite) image, then run `lsusb -t`, or you find it on the internet)
2. Edit `build.sh`, add a line like `INSTALL_MODULES="$INSTALL_MODULES kernel/drivers/net/usb/r8152.ko"` after the first definition of `INSTALL_MODULES`
3. Add a line like `drivers_to_load=r8152` to `installer-config.txt`

closed 2016-11-19 by BenjaminHae

Mausy5043, 2016-11-19:
Thanks @BenjaminHae and @kpfleming for tracking this down.