debian-pi/raspbian-ua-netinst #439: Can’t copy image to another SD card as raspbian uses whole SD card by default

palatia, 2016-12-04:
SD card comes usually with different sizes of a few bytes, so using an SD copy program don’t allow you to copy the image to another SD card. It might better to not take the always the full available bytes per SD card. An better option might be to take a bit less as available like 7.8 GB or 15.8 GB) … sure a few bytes are lost but you won’t run into issues using an SD image copy program.

Mausy5043, 2016-12-04:
You can use the parameter `rootsize` to restrict the size of the main partition.

ThomDietrich, 2016-12-04:
I think the idea is to have raspbian-ua-netinst reserve a few MBs per default to tackle this very real problem [1](https://www.reddit.com/r/RetroPie/comments/50rdmb/new_32gb_sd_card_slightly_smaller_than_the_card_i/), [2](http://raspberrypi.stackexchange.com/questions/5440/copy-existing-raspbian-installation-to-smaller-sd-card), [3](https://www.raspberrypi.org/forums/viewtopic.php?t=19468). I would support the idea.

The open question seems to be: How much is "a few MBs"? [Here](http://unix.stackexchange.com/questions/144411/dd-to-clone-to-smaller-sdcard) one user reported 44 MBytes.

Mausy5043, 2016-12-04:
Sorry for being so curtly earlier. I'm having a Dr. Who marathon with my son. 😄 

You can set a specific value for the size of the rootpartition `/` and the bootpartition `/boot`.

The size of the bootpartition is already set in the installer (currently at 128M). It should suffice for most people.
The  default behaviour of the installer is not to set the size of the rootpartition. This will result in the installer grabbing the entire remainder of the SD-card for `/`.  
We do not intend to change the default behaviour of the installer as that might break stuff for other users.
If you do not want the installer to grab the entire SD-card, then you can specify the size by adding this line in `installer-config.txt`. For example:

```
rootsize=+8000M
```

This will claim 8000 MB for the root-partition. Depending on the range of SD-card sizes that you intend to use -- and allowing for the size of the boot-partition -- the exact value you need to enter there will vary. Some experimentation is required.  

Hope this helps.

ThomDietrich, 2016-12-04:
No problem ;) All that was more or less clear. Going beyond that, the question was if you would consider to either leave a few MBs at the end of the sd card untouched by default OR introduce a install-config.txt option like `unused-after-root=50M`.

The whole point is that 50MB more or less will not make a difference in production but ease the transition to a new sd card (if that's needed by the user at some point). Yes there are tools like [this](https://github.com/Drewsif/PiShrink) or [this](http://sirlagz.net/2013/03/10/script-automatic-rpi-image-downsizer/) but having the slightly smaller partition to begin with would be handy...

Totally up to you.

kpfleming, 2016-12-05:
Seems like a reasonable proposal, and like most open source projects patches (or in this pull requests) are definitely welcome :-)

ThomDietrich, 2016-12-06:
@palatia would you like to look into that? :)

Mausy5043, 2016-12-06:
I'm still trying to understand why the parameter `rootsize` does not provide a solution for you.
It allows you to fix the size of the `/` partition during the installation. 

Further, copying SD-cards with an **installed** image does not seem to be a common requirement. For those that do require it, they are likely to use batches of the same cards. So, this solution should work fine.

diederikdehaas, 2016-12-06:
> a install-config.txt option like unused-after-root=50M

While it may sound easy, it really isn't (to implement).
See the issues regarding optimizing for speed (alignment) and changing the boot-offset. 
The trick that was used there, creating a temp partition and deleting it afterwards, can't be used in this case and you'd have to calculate the exact sector on which to end the root partition.

> 50MB more or less will not make a difference

As usual, that depends. On a 128GB sd card, yeah, who cares about 50MB. But on a 512MB card, that would be 10% and that is significant. And that's besides the implementation problem I described above.

I think that using an image-copy program or `dd` is the wrong tool for the job here.
If you'd use `rsync` you wouldn't have this issue and thus seems more fitted.

If you describe in detail the exact problem/use case, we may be able to find something better.

Mausy5043, 2017-03-19:
3+ months no update and unlikely that this will lead to changes in the installer.
Closing the issue.

If you feel the closure was in error, please re-open and add additional information.

closed 2017-03-19 by Mausy5043