debian-pi/raspbian-ua-netinst #440: ln: failed to create hard link

Mausy5043, 2016-12-11:
I get this error during the installation and sofar I've always ignored it.
```
Processing triggers for initramfs-tools (0.120+deb8u2) ...
ln: failed to create hard link ‘/boot/initrd.img-4.4.0-1-rpi.dpkg-bak’ => ‘/boot/initrd.img-4.4.0-1-rpi’: Operation not permitted
update-initramfs: Generating /boot/initrd.img-4.4.0-1-rpi
```

However, when using `apt-get` I frequently also see this error. It's starting to annoy me. What causes this? And what can/must be done to get rid of it?

goranche, 2016-12-11:
`/boot` is a FAT filesystem, which doesn't support hard links

Mausy5043, 2016-12-12:
ofcourse.
😞

closed 2016-12-12 by Mausy5043

diederikdehaas, 2016-12-13:
> the /boot partition has to be a FAT filesystem

AFAIK the first partition must be FAT and must contain the firmware/bootloader files.
http://sjoerd.luon.net/posts/2015/02/debian-jessie-on-rpi2/ had it mounted on `/boot/firmware` I believe which makes me think that *technically* you can have `/boot/` as ext4 (for example) and have `/boot/firmware` be the first partition with FAT. The *practical* problem is that (currently) the package that provides the firmware/bootloader files places them in `/boot/` and switching that in Raspbian will cause (a lot of) problems.

But ... there is https://wiki.debian.org/RaspberryPi3 and the firmware (for RPi3) will get placed in `/boot/firmware`, see the [postinstall of that package](https://anonscm.debian.org/cgit/pkg-raspi/raspi3-firmware.git/tree/debian/raspi3-firmware.postinst)

goranche, 2016-12-14:
in that case I think you'd have to put the booting kernel on that partition as well, so you wouldn't accomplish anything, really...