debian-pi/raspbian-ua-netinst #441: Yet another issue of RPi3

mytrain, 2016-12-18:
Hi

Using the script with the latest version of the kernel (2016-11-25-raspbian-jessie-lite.img) my RPi3 is not booting anymore (used to work with 2016-03-18-raspbian-jessie-lite.img).

The Foundation introduced a new file bcm2710-rpi-cm3.dtb. Is this messing up the boot ?

Thanks

Mausy5043, 2016-12-19:
Please provide more context. See [here](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md) for directions.

Could you please also share your `installer-config.txt`, `post-install.txt` and either the log-file or the error message that appears on the screen?

mytrain, 2016-12-19:
In fact I'm running modified version of rcS that aims at allowing updating of my RPi from an ISO file zipped on an USB drive (instead of doing a full network install has this project does). I also integrated websocketd that allow to run a mini web/websocket server in order to allow the user to connect via web browser (and a cable to a static IPIPA address 169.254.254.254)

I was running the old 1.0.8.1 version

closed 2016-12-19 by mytrain