debian-pi/raspbian-ua-netinst #443: cdeboostrap-static using local repos

awelters, 2016-12-23:
I would like to not use the internet for package installation, is this possible?  If so, how to go about doing this?

Mausy5043, 2016-12-24:
I believe this requires you to specify your own mirror. In the `installer-config.txt` set `mirror=http://my.server.lan` where "my.server.lan" is the FQDN of your local mirror. 

This is assuming you already have the required infrastructure on your local LAN.

goranche, 2016-12-24:
or create a "local" (on the SD card) apt mirror (with apt-mirror), and access it by passing a file url as the mirror... `mirror=file:///boot/apt-mirror`

but you'll have to figure out what packages to add to the mirror, and don't forget that the boot partition will need to be quite a bit larger to hold the mirrored files

awelters, 2016-12-24:
Any way you can describe adjusting the sources list to just pull locally?  Or what if I used multistrap to build my my chrootable /rootfs, how would I go about replacing the line where cdebootstrap-static does that?  I realize that's kinda the whole point of this raspbian-ua-netinst, but everything else about building a debian pi in this repo works swimmingly, just getting really stuck avoiding the use of cdebootstrap if I can or not having it go to the internet.  Thanks.

goranche, 2016-12-25:
this is something that'll require quite a bit of change in the installer itself, it's not something that the installer was intended for, mind you... 😄 

but basically, what you'd do is change the [default apt sources setup](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/scripts/etc/init.d/rcS#L593-L598) and only add a local source to the `/etc/apt/sources.list` file (on the `/rootfs` filesystem, of course).

There is no way of "plugging in to" the installer process before it starts it's bootstrapping of the system, so this would have to be changed in the installer script itself, and you'd have to build your own installer image. Debugging this will be a pain.

After setting up the sources, you'll have to make sure the local "repository" will be available in the chroot environment of the bootstrapped system, which pretty much means doing a simple bind mount.

That's basically it, in a (very small) nutshell... information on specifics shouldn't be hard to find, but again (and I can't stress this enough), this isn't something that's built into the installer and will probably result in a lot of pulled out hair and broken keyboards...
I'd really only recommend this for someone that's really comfortable with (or at least knows how to search for the needed information) how apt works, is at home in the command line and has an intricate knowledge of linux filesystems (linux in general, I guess)...

A good place to find information on how this will be performed is installer images (the ones that we used to burn to CDs, back when the world still was black and white), look at how it's done there...
They will probably use [apt-cdrom](https://manpages.debian.org/cgi-bin/man.cgi?sektion=8&query=apt-cdrom&apropos=0&manpath=sid&locale=en), but it's a start...

Oh, and don't forget that you also have to create a local apt mirror holding all the packages that you'll need for installation...

Again, it's not small task, so I'd really think about just doing a networked install and moving the installed image to where you need it (and don't have access to a network)

Mausy5043, 2017-10-17:
No activity on this issue for almost a year. Closing.
If you feel the closure was in error please feel free to re-open and add additional info.

closed 2017-10-17 by Mausy5043