debian-pi/raspbian-ua-netinst #444: no sound card installed (aplay: device_list:268: no sound  card found ...)

dirkclemens, 2016-12-27:
I installed jessie on a RPi2 using ua-netinst and it worked fine besides I do no longer have sound via the 3.5 jack.
I tried everything google told me, e.g.
* sudo modprobe snd_2835
* /ect/modules (snd_bcm2835) -->this is wrong: snd-bcm2835
* sudo amixer cset numid=3 1
* /boot/config.txt -> dtsparam=audio=on (and hdmi_drive=2 and enable_uart=1 and ... in all combinations)
* ~/.asoundrc (pcm.!default { type hw card 0 } ctl.!default { type hw card 0})
* dito in /etc/asound.conf
* and lots more, all without success.

* aplay -l -> aplay: device_list:268: no sound  card found ...
* cat /proc/asound/cards -> "--- no soundcards ---"
* uname -a -> Linux smarthome 4.4.0-1-rpi2 #1 SMP Debian 4.4.6-1+rpi14 (2016-05-05) armv7l GNU/Linux
* amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 58983 [90%] [on]
  Front Right: Playback 58983 [90%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 65536 [100%] [on]
  Front Right: Capture 65536 [100%] [on]

Any hints?

Mausy5043, 2016-12-28:
As we disabled sound on the default installation you would need to include this in the `post-install.txt` to re-enable it:
```
echo "snd-bcm2835" >> /rootfs/etc/modules
```
OR  
you could `sudo modprobe snd-bcm2835` on the installed system.

Then go from there.

We already discussed this a bit in #393. 
Does that discussion answer your question?

dirkclemens, 2016-12-28:
thanks for your response, but as written above, I already did this without success.
btw: the module is named snd_bcm2835 and not snd-bcm2835 ("_" instead of "-")

Mausy5043, 2016-12-28:
Both "-" and "_" seem to work for me.

Since sound is not supported out-of-the-box I'm guessing you'll need to install some additional packages. Can't help you there though.

Mausy5043, 2016-12-28:
BTW, you said:
> I do no longer have sound via the 3.5 jack

What was your previous set-up? The one that did work.

dirkclemens, 2016-12-28:
my old setup broke because the sd-card was corrupted. now i've installed everything to a usb-harddrive (160GB) and started from scratch with ua-netinst. I do have backups, but I suppose that the problem is not having something misconfigurated but a missing driver/module/additional packages, as you already mentioned.

Mausy5043, 2016-12-28:
Was that on a stock Raspbian image or raspbian-ua-netinst?

dirkclemens, 2016-12-28:
sorry, i forgot: it was an original wheezy image. but an updated on (latest firmware).

Mausy5043, 2016-12-28:
> sorry, i forgot: it was an original wheezy image. but an updated on (latest firmware).

Ah well, that explains it. The RPF images have sound support installed by default. Systems installed with this repo don't.

aristosv, 2017-01-04:
I had the exact same issue. Only I was trying to play audio with mpg123. No matter how many suggestions I tried, audio would not work. But for some weird reason omxplayer could play audio. So I just changed my scripts to use omxplayer.

oupala, 2017-06-04:
I think this is due to the new way raspbian is handling the devicetree.

To enable the soundcard, you just have to add a line to a config file:

Add `dtparam=audio=on` to the file `/boot/config.txt`, then restart the server.

Mausy5043, 2017-10-17:
Closing this issue because it is not our intention to provide sound support by default.
You may use the `post-install.txt` mechanism to make the appropriate changes.
If you feel the closure was in error please feel free te re-open and add your new information.

closed 2017-10-17 by Mausy5043