debian-pi/raspbian-ua-netinst #445: Permission Denied in ssh while running rasbian in headless mode

sankarhari, 2017-01-01:
I try to connect my raspberry pi with my laptop runs on linux mint. I connect to my laptop directly with a Ethernet ports when a try to login in to raspbian jessi with ssh i am getting permission denied error message. Followed all steps properly which is given in guide. 

What to do?

goranche, 2017-01-01:
could you please provide the info as [described here](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md#reporting-issues)?

your `installer-config.txt` and `post-install.txt` files will probably be most important in this case.

Mausy5043, 2017-01-01:
Please also provide the output of `ssh -vvv pi@raspberry.local` 
(replace `pi` and `raspberry.local` with the appropriate username and hostname)

Mausy5043, 2017-01-01:
For future reference I'm putting this here: 

In theory, using SSH to connect to a Raspberry Pi with a raspbian-ua-netinst installed system, there are only four ways that things might not be working as expected. This is ofcourse **provided** that you are using the installer in the intended way and you have not modified the SSH installation.

Let's say you are trying to log into a default installation. This has user `root` set-up with password `raspbian` and a hostname `pi` (you are strongly encouraged to change these defaults, but I'll use them for the sake of this discussion). The normal way would be to log in using `ssh root@pi` optionally adding a domain e.g. `ssh root@pi.local`. 

## How to fail to log in using SSH:
1. _you type the hostname/domainname wrong_. 
Let's say you typed `py` instead of  "pi":
```
$ ssh root@py
ssh: Could not resolve hostname py: nodename nor servname provided, or not known
```
The hostname can't be resolved so SSH won't be able to make a connection. The errormessage you get is very straightforward.

2. _you type in the wrong username_. 
Let's try to use `toor` instead of "root" here:
```
$ ssh toor@pi
The authenticity of host 'pi (10.0.1.11)' can't be established.
ECDSA key fingerprint is SHA256:5nI4y57eGopzov8iF3Bk19gSQJVlb5gUXXXXXXXXXXXXX.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'pi' (ECDSA) to the list of known hosts.
toor@pi's password:
Permission denied, please try again.
toor@pi's password:
Permission denied, please try again.
toor@pi's password:
Permission denied (publickey,password).
```
The first four lines are what you get when you try to connect to a new host for the first time. So, at the start it is suggested that everything is okay. Then the host can't find the user "toor". But it does not let on to this for security reasons. You will be able to **try** logging in as this non-existing user. After three times SSH gives up.

3. _you get the user and hostname right, but you type the wrong password_. 
The default password is `raspbian`; but typing `rasbian` is a common error (as was shown in the original description of this issue). 
```
$ ssh root@pi
root@pi's password:
Permission denied, please try again.
root@pi's password:
Permission denied, please try again.
root@pi's password:
Permission denied (publickey,password).
```

4. _The private/public key is missing or corrupt_.
If there is no valid private/public key installed for the user, the login procedure will automatically default to the normal login sequence; asking you for the user's password and the above three options become available.
At some point during or after the installation you may have provided a secure key. My favorite method is to do this after the installed system is complete. I make sure that the directory `/home/<user>/.ssh` exists (where <user> is the name of the user (if not root)) or `/root/.ssh` and that it has `0700` permission.
Then, I copy my public key to the Pi with: 
```
$ scp .ssh/id_rsa.pub root@pi:/root/.ssh/authorized_keys
The authenticity of host 'pi (10.0.1.11)' can't be established.
ECDSA key fingerprint is SHA256:5nI4y57eGopzov8iF3Bk19gSQJVlb5gUXXXXXXX.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'pi,10.0.1.11' (ECDSA) to the list of known hosts.
root@pi's password:
```
Provided the correct password was entered, this should work flawlessly. And a subsequent `ssh root@pi` will result in you getting logged in without any additional interaction.
However, let's say that the key got corrupted. In that case the login procedure will again default to asking you for the password.

diederikdehaas, 2017-03-19:
No response for 2.5 months, therefor closing the issue.
If you do have the requested info, feel free to reopen it.

closed 2017-03-19 by diederikdehaas

SISYPHE2, 2018-03-28:
did you write sudo in front of the command?

pkese, 2018-12-15:
In my case, I've set my password directly on RPi not knowing that on raspberry's default locale (en-UK) some keys would map into different characters than on my main computer with en-US locale.

So I've booted the RPi again, reset my password to something really simple (only ascii letters) and then it worked.

shriomkalra, 2019-04-22:
ssh -v pi@192.168.0.103
OpenSSH_7.9p1 Debian-9, OpenSSL 1.1.1b  26 Feb 2019
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to 192.168.0.103 [192.168.0.103] port 22.
debug1: Connection established.
debug1: identity file /root/.ssh/id_rsa type -1
debug1: identity file /root/.ssh/id_rsa-cert type -1
debug1: identity file /root/.ssh/id_dsa type -1
debug1: identity file /root/.ssh/id_dsa-cert type -1
debug1: identity file /root/.ssh/id_ecdsa type -1
debug1: identity file /root/.ssh/id_ecdsa-cert type -1
debug1: identity file /root/.ssh/id_ed25519 type -1
debug1: identity file /root/.ssh/id_ed25519-cert type -1
debug1: identity file /root/.ssh/id_xmss type -1
debug1: identity file /root/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_7.9p1 Debian-9
debug1: Remote protocol version 2.0, remote software version OpenSSH_7.9p1 Debian-5
debug1: match: OpenSSH_7.9p1 Debian-5 pat OpenSSH* compat 0x04000000
debug1: Authenticating to 192.168.0.103:22 as 'pi'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ecdsa-sha2-nistp256 SHA256:jrKSlTuGQ0Htl35K17txt2hCiAhHC6vBoaKVOou90K0
debug1: Host '192.168.0.103' is known and matches the ECDSA host key.
debug1: Found key in /root/.ssh/known_hosts:1
debug1: rekey after 134217728 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey after 134217728 blocks
debug1: Will attempt key: /root/.ssh/id_rsa 
debug1: Will attempt key: /root/.ssh/id_dsa 
debug1: Will attempt key: /root/.ssh/id_ecdsa 
debug1: Will attempt key: /root/.ssh/id_ed25519 
debug1: Will attempt key: /root/.ssh/id_xmss 
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,ssh-rsa,rsa-sha2-256,rsa-sha2-512,ssh-dss,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,password
debug1: Next authentication method: publickey
debug1: Trying private key: /root/.ssh/id_rsa
debug1: Trying private key: /root/.ssh/id_dsa
debug1: Trying private key: /root/.ssh/id_ecdsa
debug1: Trying private key: /root/.ssh/id_ed25519
debug1: Trying private key: /root/.ssh/id_xmss
debug1: Next authentication method: password
pi@192.168.0.103's password: 
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
pi@192.168.0.103's password: 

after using 'toor' as password. Please help out!

Mausy5043, 2019-04-28:
> The default password is `raspbian`

npacifique, 2019-05-10:
> In my case, I've set my password directly on RPi not knowing that on raspberry's default locale (en-UK) some keys would map into different characters than on my main computer with en-US locale.
>
> So [...]
thank you @pkese , I spent time thinking that there is something wrong with my ssh config, but after reading your post, I used nano to view my password in clear text then I realized my ssh client computer was using a different keyboard layout.

zoroloco, 2019-10-14:
Thank you PKSE.  The password I initially set directly was different locale from the remote pc.

cmartinez-effectual, 2021-04-19:
> In my case, I've set my password directly on RPi not knowing that on raspberry's default locale (en-UK) some keys would map into different characters than on my main computer with en-US locale.
> So [...]

This was my problem twice in a row, because I was too stupid to remember it the second time it happened to me. So I will leave this comment here to remind myself that if you set the password locally change the keyboard layout to US.