debian-pi/raspbian-ua-netinst #446: Doesn't work with Raspberry Pi Model 2 B v1.2 (Revision a22042)

awelters, 2017-01-04:
So the latest version from raspbian:

diederikdehaas, 2017-01-05:
> Is it possible the linux-image needs to be upgraded from 4.4.0-1

Could be. Keep in mind that Debian/Raspbian uses ABI revision numbers, not the minor versions from upstream linux as raspberrypi.org does.

Can you first try using newer firmware files? 
Your issue seems similar to https://github.com/debian-pi/raspbian-ua-netinst/issues/415 for the Pi 1B+ rev 1.2
To do so, grab the *.dat, *.elf and bootcode.bin and potentially bcm2709-rpi-2-b.dtb from https://github.com/raspberrypi/firmware/tree/master/boot and use those files to overwrite the ones you placed on the sd card and report back.

awelters, 2017-01-05:
I get passed the rainbow screen and it tells me that it can't run "/etc/init.d/rcS" because permission denied.  So what does this mean?

awelters, 2017-01-05:
It appears to have fixed the issue.

awelters, 2017-01-06:
So now that I did that everything works fine until you try to unmount at the very very end of the rcS script and it fails to unmount, reboots, but rainbow screens again.  On the sd card I can see the following files:

config-4.4.0-1-rp2 
initrd.img-4.4.0-1-rp2
System-map-*
vmlinuz-*

My guess is even though I copied over those updated firmware files, the script process adds them right back and everything blows up.

Suggestions?

diederikdehaas, 2017-01-06:
The installer installs the firmware files (package) which doesn't work for your Pi model. Copying the same files onto the installed system likely makes it boot/work again.
What I wanted to figure out with this test is whether an updated firmware package would be all that was needed and it appears so.

What does confuse me though is the permission denied msg on `/etc/init.d/rcS` you got and the unmount failure

awelters, 2017-01-06:
The issue with permissions on the /etc/init.d/rcS was that I didn't make it executable on my build system aka I needed to do a 'chmod 751' on that file before it got packaged up.

awelters, 2017-01-06:
I was able to use the following fork and it worked: https://github.com/FooDeas/raspberrypi-ua-netinst

Might I suggest you use their technique to make sure to have the latest firmware.  I could not get yours to work.

Mausy5043, 2017-01-07:
Merge of #447 should resolve this issue. Closing.

If you feel the issue remains, please feel free to add additional info and reopen this issue.

diederikdehaas, 2017-01-07:
It helps, but it's not enough. I asked for an updated firmware package and when that's available then this issue can be closed (would need a rebuild of the installer though).

diederikdehaas, 2017-01-21:
Asked for an updated firmware package on IRC, but no result so far and have therefor reported it upstream at https://bugs.launchpad.net/raspbian/+bug/1658340

Mausy5043, 2017-12-11:
@awelters : Please confirm this now works with the beta3 release?

Mausy5043, 2017-12-18:
Assuming that, based on [this line](https://github.com/debian-pi/raspbian-ua-netinst/blob/6f17a440b8fa8a30c1538de14171a6a453ff0a7d/scripts/etc/init.d/rcS#L208), the Pi Zero W will boot with the current `beta3` release I'm closing this issue.
If you feel the issue was closed in error, please don't hesitate to re-open and add new information.

closed 2017-12-18 by Mausy5043