debian-pi/raspbian-ua-netinst #450: Dynamic /dev/loop support

richard-scott, 2017-01-30:
Hi,

I have an Ubuntu system that has the live kernel patching activated. This process consumes 3 loop devices and stops the 'buildroot.sh' script from working as this depends on /dev/loop0 which is currently in use.

This [PR](https://github.com/debian-pi/raspbian-ua-netinst/pull/431) will detect the next free loop device and use that instead of being fixed to /dev/loop0.

Thanks,

Rich

goranche, 2017-01-30:
hey, @richard-scott 
thanks for your contribution... and yes, it's true the loop0 is fixed, and "probably" should be changed dynamically 😇 

we'll try it out and merge if all works out 👍

diederikdehaas, 2017-03-19:
With the merge of #431 this issue can be closed

closed 2017-03-19 by diederikdehaas