debian-pi/raspbian-ua-netinst #451: Stretch Support [Guide]

jrwr, 2017-02-11:
Here is what I did to get Stretch working with ua-netinst, Ran will full CDE Debug to make sure everything went well, the curl shim was needed because cdebootstrap requires curl and including it was too much work when busybox wget was fine. Tested on a rPi B+ v1 (512MB) with the 1.1.x branch

- Still Built the base image as Jessie
- Uploaded cdebootstrap to cdebootstrap-static_0.7.6_armhf.deb
- edited build.sh to include a curl shim into /bin/curl
`#!/bin/busybox sh
while [[ $# -gt 1 ]]
do
key="$1"

case $key in
    --output)
    OUTPUT="$2"
    shift # past argument
    ;;
    *)
            # unknown option
    ;;
esac
shift # past argument or value
done
/bin/busybox wget --output "${OUTPUT}" $1
`
- edited build.sh to always include systemd

jrwr, 2017-02-11:
I have plans to test for some failure modes in the curl shim for cdebootstrap but using just a normal install it worked well.

diederikdehaas, 2017-02-11:
>  full CDE Debug

= `cdebootstap --debug` ?

> curl shim was needed because cdebootstrap requires curl and including it was too much work when busybox wget was fine.

You could consider using busybox from stretch too, which no longer prefers applets:
```
busybox (1:1.22.0-19) unstable; urgency=medium

  * busybox-udeb, udhcpc: Remove /udhcpc/usr/share/udhcpc/default.script,
    erroneously included in version 1:1.22.0-18
  * Disable CONFIG_FEATURE_PREFER_APPLETS in standard build (Closes: #821307)

 -- Ben Hutchings <ben@decadent.org.uk>  Sun, 17 Apr 2016 16:37:24 +0100
```
[source](https://tracker.debian.org/news/762568)
but you could run into other issues though ... (haven't tried it myself yet)

jrwr, 2017-02-11:
Yep, I ran it with --debug to see the output from wget, found a issue where it installs a really old kernel, doing some package overrides now.

diederikdehaas, 2017-02-11:
Which kernel version? 
Are you trying to install a kernel headers or dkms package? See https://bugs.launchpad.net/raspbian/+bug/1532559 for that then

jrwr, 2017-02-11:
it had installed something like 3.19. I'm forcing 4.4 with a packages= config line to see what it does right now

diederikdehaas, 2017-02-11:
Ah yeah, crap. I did report it to plugwash **quite** some time ago, but apparently he didn't fix it :-/

jrwr, 2017-02-11:
Blah, Also holy hell is there a ton of hard coded paths in build, would be nice to be a little more flexible with it as Debian changes ( I had gone down a path to try and include curl and all its libs and promptly gave up on that)

diederikdehaas, 2017-02-11:
Do an `aptitude search linux-image` on a stretch system and you'll likely won't find the 4.4 kernel.
If that is indeed the case, you need to add jessie sources.

> holy hell is there a ton of hard coded paths in build

could you report (in detail) each an every issue you encounter? Would help us in getting stretch support ootb working.

jrwr, 2017-02-11:
Sure thing, want them as separate issues?

diederikdehaas, 2017-02-11:
nah, dump them all here ;-)

jrwr, 2017-02-11:
So, besides cdebootstrap being buggy in the older versions, the curl shim was the bigger issue, Im doing another test install now with the force 4.4 package (It is in the release, I had to grep the packages list 👍 )

jrwr, 2017-02-11:
We really should look into porting debian installer outright to rPi, Wonder how that would go

diederikdehaas, 2017-02-11:
That would be a MAJOR task and it would be hard to support the features the installer now has. 
The first attempt at doing that just died at some point. Dunno whether that's because it was too hard or the author lost interest though. See https://raspbian.org/RaspbianInstaller

jrwr, 2017-02-11:
I figured as much, looking at even how complex debootstrap is, its no surprise.  I did notice that the rPi Foundation repo does not support stretch yet.

jrwr, 2017-02-11:
Oh well thats interesting, it didn't install init at all -- looks like it skipped over systemd

jrwr, 2017-02-11:
kernel=vmlinuz-3.18.0-trunk-rpi found in the config.txt - forcing it to the 4.4 kernel it installed

jrwr, 2017-02-11:
wont boot with 4.4, sitting at rainbow. Interesting.

diederikdehaas, 2017-02-11:
That's strange. There are 1 or 2 newer RPi revisions that are not supported by the firmware, but yours shouldn't be a problem.
Have you checked your `/boot/config.txt` file if it may be pointing to the wrong kernel?

jrwr, 2017-02-11:
So it booted with 3.14 kernel, just no init, with the extra 4.4 install it just sat there at rainbow

I noticed that this line:
`KERNEL_VERSION=$(sed -n -e "/^Package: ${kernel_meta_package}$/,/Package: /{s/^Depends: .*linux-image-\(.*-rpi2\{0,1\}\).*$/\1/p}" /rootfs/var/lib/dpkg/status)` in build might be at fault for selecting the wrong kernel

diederikdehaas, 2017-02-11:
You should see that then in the installer log file.
Normally it installs a kernel meta package for your Pi version, which in turn installs the real kernel.
But no `init` sounds really strange ...

jrwr, 2017-02-11:
Package: linux-image-rpi
Source: linux-latest (63+rpi4)
Version: 3.16+63+rpi4
Architecture: armhf

Found in the Packages list forStretch

jrwr, 2017-02-11:
Im looking at all the packages around linux-* and most are pointing to version 3.10~ of the kernel

diederikdehaas, 2017-02-11:
That is REALLY strange. I first though you made a typo and actually meant 3.18.
For the RPi 1, it should install `linux-image-rpi-rpfv` ([source](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L468))

jrwr, 2017-02-11:
it was 3.18, but 3.18 was before systemd support I think, thats why init broke
I'm going to make a mod to rcS to prevent linux-image-rpi-rpfv being installed. I think its confusing dpkg

jrwr, 2017-02-11:
just did a quick edit to change kernel_meta_package to linux-image-4.4.0-1-rpi and linux-image-4.4.0-1-rpi2

diederikdehaas, 2017-02-11:
> 3.18 was before systemd support I think, thats why init broke

nope, it has to be sth else.

> I'm going to make a mod to rcS to prevent linux-image-rpi-rpfv being installed. I think its confusing dpkg

That sounds unlikely to me. 

What you should check is whether the `init` package is installed. That in turn installs systemd by default.
What would help is if you can share log files (make sure passwords are blanked out) so I/we don't have to speculate, but can investigate.

jrwr, 2017-02-11:
Ya, Im just using defaults on the installer for testing, Im on a windows host so ext4 is kinda hard, If this install fails, Ill get you those logs

jrwr, 2017-02-11:
found the issue:
Adding boot config for kernel version: ''
Copying kernel to /boot/kernel.img... cp: can't stat '/rootfs/boot/vmlinuz-': No such file or directory
FAILED !
Creating default cmdline.txt... OK

[The full log is here](https://hastebin.com/umibupakas.log)

Its a uncaught error in the rcS at the end of the install, I'm making a full disk image for inspection now

diederikdehaas, 2017-02-11:
That means that it didn't have/find the kernel meta package, which in turn make other things fail. 
And that `cp` statement should check whether the source actually exists :-P

jrwr, 2017-02-11:
so it all stems back to the meta packages being broke :3

jrwr, 2017-02-11:
so trying with the installer's kernel.img in the config fails to boot with the 4.4's intramfs image, just sits at rainbow

diederikdehaas, 2017-02-11:
> so it all stems back to the meta packages being broke :3

You can say that again, omg:
```
diederik@rpi2-stretch:~$ aptitude show linux-image-rpi2-rpfv                                                                                                                         
Package: linux-image-rpi2-rpfv                                                                                                                                                     
Version: 4.4+63+rpi5                                                                                                                                                               
State: installed                                                                                                                                                                   
Automatically installed: no                                                                                                                                                        
Priority: optional                                                                                                                                                                 
Section: kernel                                                                                                                                                                    
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Architecture: armhf
Uncompressed Size: 29,7 k
Depends: linux-image-4.4.0-1-rpi2
Description: This metapackage will pull in the raspbian kernel for the raspberry pi 2
 based on the version currently reccomended by the raspberry pi foundation (currently 3.18).

diederik@rpi2-stretch:~$ aptitude show linux-image-rpi-rpfv
Package: linux-image-rpi-rpfv            
Version: 3.18+63+rpi4
State: installed
Automatically installed: no
Priority: optional
Section: kernel
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Architecture: armhf
Uncompressed Size: 29,7 k
Depends: linux-image-3.18.0-trunk-rpi
Description: This metapackage will pull in the raspbian kernel for the raspberry pi 1
 based on the version currently reccomended by the raspberry pi foundation (currently 3.18).
```

diederikdehaas, 2017-02-11:
And IIRC 3.18 does indeed not work with the (latest) firmware version

jrwr, 2017-02-11:
Yep, Kernel is refusing to load the intrd image guess its just not finding it

diederikdehaas, 2017-02-11:
can you post/share your `/boot/config.txt`?

jrwr, 2017-02-11:
Sorry, I had posted a snippet of it earlier
```

[pi1]
kernel=kernel-rpi1_install.img
initramfs initrd.img-4.4.0-1-rpi
boot_delay=1
[pi0]
kernel=kernel-rpi1_install.img
initramfs initrd.img-4.4.0-1-rpi
boot_delay=1

```

jrwr, 2017-02-11:
[Here is the Content of the entire /boot](http://jrwr.io/E.zip)

jrwr, 2017-02-11:
I should note that I have gotta a dist-upgrade from a normal jessie install to work on a pi1

diederikdehaas, 2017-02-11:
That kernel and initramfs line does not seem correct.
Here's what I have:
```
$ cat /boot/config.txt 
[pi1]
kernel=vmlinuz-3.18.0-trunk-rpi
initramfs initrd.img-3.18.0-trunk-rpi followkernel
# to enable DeviceTree, comment out the next line 
#device_tree=
gpu_mem=32M
[pi2]
kernel=vmlinuz-4.4.0-1-rpi2
initramfs initrd.img-4.4.0-1-rpi2 followkernel
#kernel=vmlinuz-4.6.0-1-armmp-lpae
#initramfs initrd.img-4.6.0-1-armmp-lpae followkernel
gpu_mem=32M
```
So yours should be like:
```
[pi1]
kernel=vmlinuz-4.4.0-1-rpi
initramfs initrd.img-4.4.0-1-rpi followkernel
boot_delay=1
[pi0]
kernel=vmlinuz-4.4.0-1-rpi
initramfs initrd.img-4.4.0-1-rpi followkernel
boot_delay=1
```

diederikdehaas, 2017-02-11:
Reported the (wrong) 3.18 kernel dependency here: https://bugs.launchpad.net/raspbian/+bug/1663944

jrwr, 2017-02-11:
Still no boot on the rPi1 or a piZero, Tested Hardware with working image onto same SD card (Jessie Install)

diederikdehaas, 2017-02-11:
And without the `boot_delay` line?

jrwr, 2017-02-11:
Still no go, there are no boot codes (green LED not blinking) as well, Just rainbow

jrwr, 2017-02-11:
thanks for all the fast responses on this, if you are at defcon this year, I owe you a few beers

diederikdehaas, 2017-02-11:
Then I'm out of ideas (for now).
So the easiest way for now is installing a minimal system with jessie and then do the dist-upgrade.

Adding support for stretch is on my radar, but first need to fix some other things :-P

jrwr, 2017-02-11:
Yep, it boots with a jessie initramfs (from my other default install from master branch)
and init is STILL missing

jrwr, 2017-02-11:
Ill be mulling over the build scripts once they fix the meta packages, Ill start doing a audit on the Stretch packages and make sure nothing else stands out as broken (Ill Compare Mainline debian's packages to raspian)

jrwr, 2017-02-11:
Im going to do some testing in a chroot and see what cdebootstrap is messing up, it looks like a package is missing or config scripts are missing something (package included scripts)

jrwr, 2017-02-12:
So I've been doing some research and found a few tools floating around, 
Hexxeh/rpi-update is really nice for installing the latest firmware / kernel outside of the current repo, no matter the dist uses a mirror off the rpi.org repo for firmware as well. im going to whip up a mod to rcS to use it for the kernel install, you would need to do it chrooted into the rootfs when you use it since all the depends it has.

diederikdehaas, 2017-02-12:
I (strongly) recommend you don't do that. The `rpi-update` tool is meant for kernel development, not for regular use. It can _literally_ break from one commit to the next.

What you can do, is use the kernel+firmware from the raspberrypi.org repository. The kernel+ firmware you can get via `rpi-update`, after stabilization and testing, is made into the kernel package from the raspberrypi.org repository.
If you go that route, you'll likely need to remove the `01vcgencmd.pref` file from apt preferences as that is specifically used to force the raspbian.org one, but the one from raspbian.org and raspberrypi.org are mutually exclusives (breaks relationship in Debian terms).

jrwr, 2017-02-12:
Gotta love the package ecosystem that rpi.org has forced on everyone :-\ So
it looks there are a few more packages with missing depends for the
initramfs building, I think its building the wrong format, Do you know what
format it should be set to (I know installer uses cpio.gz style)

diederikdehaas, 2017-02-12:
Yeah, the situation is far from ideal :-/
Forcing the raspbian packages was specifically done as the rpi.org package were substandard (imo), but it has improved since (a bit).

If you provide _exact_ error messages, I may be able to help ...
Right now it's only "it doesn't work" to me, which is not much of a starting point ;-)

jrwr, 2017-02-12:
Im working on that, its the main installer not including init AND building
the initramfs correctly, Ill do some more digging if you force it to use
stretch

ralin3, 2017-08-18:
Any updates to this, now that raspbian is also out in stretch?

Mausy5043, 2017-09-10:
I'd advise to keep using `jessie`. For a while at least. Migrating the installer *and* the installed system isn't an easy task. It'll certainly require us to be in a stable v1.1.x release.

You could try to upgrade to `stretch` using [these instructions](https://linuxconfig.org/raspbian-gnu-linux-upgrade-from-jessie-to-raspbian-stretch-9) which might be executed in the `post-install.txt` (not tested, so YMMV!).

Mausy5043, 2017-09-21:
[This PR](https://github.com/FooDeas/raspberrypi-ua-netinst/pull/90) might help

ralin3, 2018-01-19:
I still get "unknown suite" (or whatever that errormessage is) when using a build from 1.1.x-branch I created a couple of days ago.

oupala, 2018-03-06:
Are we close to a stable release of v1.1.x installer?

If yes, it could be time to work on installing current stable of raspbian...

ralin3, 2018-03-07:
I don't know if there's much activity here at all unfortunately...
v1.1.x works just fine, I've based my builds on it for a long time, but stretch-support is lacking.

goranche, 2018-03-08:
I'll try to go through the issues, and see what has been going on and where things stand...

always happy to see PRs, of course 😉

[3 more replies]

closed 2019-12-11 by diederikdehaas