debian-pi/raspbian-ua-netinst #452: rpi-update rendering system unusable

ThomDietrich, 2017-02-26:
Hello,
lately `rpi-update` will stop a raspbian-ua-netinst (v1.1) system from rebooting. I do not have more information at this point but multiple users are already effected. Are you aware of that problem?

diederikdehaas, 2017-02-26:
> Are you aware of that problem?

No. But each and every time `rpi-update` comes up I tell ppl to stay away from it as it is not meant for normal people to use. Now you know why.
References:
https://github.com/debian-pi/raspbian-ua-netinst/issues/225#issuecomment-100497541
https://github.com/debian-pi/raspbian-ua-netinst/issues/435#issuecomment-256940661
https://github.com/raspberrypi/documentation/commit/a5c386cb3691fd9cfea0f99f303db023ff59dff1

And luckily, after much nagging from me and others, there's now even a note on the official repo: https://github.com/Hexxeh/rpi-update#notes

I wouldn't be surprised if the issue is related to kernel 4.9.
This installer installs a different kernel and uses lines in `/boot/config.txt` to point to those. I don't know whether you use sth like that, but after installing a new kernel (version), those lines need to be updated.

ThomDietrich, 2017-02-28:
Hey @diederikdehaas, thanks for confirming what I've already guessed. Also thanks for pointing me at the "notes" section. `rpi-update` is now discouraged in our project.

closed 2017-02-28 by ThomDietrich