debian-pi/raspbian-ua-netinst #454: Will there be a new release available soon?

CWempe, 2017-03-14:
I got a question: Will there be a new release available soon?

I just discovered this great project and want to use it.

I saw there are already many issues fixed in different branches.
(Like creating a new user with sudo and defining locale-settings in the `installer-config.txt`.)
But I cannot use this, because there is new new release for 1.1.x.

btw: Is this the right place to ask these kind of questions? I could not find a forum.

Mausy5043, 2017-03-19:
I know @diederikdehaas was preparing that. But there where a couple of changes needed before the release could be made. I forget which.

CWempe, 2017-03-19:
Ok.

I was just afraid, this was forgotten or the project had become abandoned. 😄

closed 2017-03-19 by CWempe