debian-pi/raspbian-ua-netinst #455: Booting on a Pi Zero W

netsgnut, 2017-03-15:
Previously I have got a microSD card with Raspbian configured on a Pi Zero. Recently with the Pi Zero W released I tried to use my old microSD on the new board, but it would seem unbootable with no indicator lights. Tried `apt-get upgrade`-`apt-get dist-upgrade`, as well as reinstalling the whole image on the old Pi Zero, but those don't seem to work.

I am still not entirely sure what could be the problem yet, but I suspect there may be a library/firmware missing, or the kernel might have modules missing.

Any ideas or suggestions? Thanks.

Below please find some basic `uname -a`, `ls -al /`, `ls -al /boot` and `cat /boot/installer-config.txt`.

```
root@raspbian:~# uname -a
Linux raspbian 4.4.0-1-rpi #1 Debian 4.4.6-1+rpi14 (2016-05-05) armv6l GNU/Linux

root@raspbian:~# ls -al /
total 88
drwxr-xr-x 21 root root  4096 Mar 15 13:46 .
drwxr-xr-x 21 root root  4096 Mar 15 13:46 ..
drwxr-xr-x  2 root root  4096 Mar 15 13:45 bin
drwxr-xr-x  4 root root 16384 Jan  1  1970 boot
drwxr-xr-x 14 root root  3120 Mar 15 16:17 dev
drwxr-xr-x 56 root root  4096 Mar 15 17:29 etc
drwxr-xr-x  2 root root  4096 Jan  7  2015 home
lrwxrwxrwx  1 root root    27 Mar 15 13:46 initrd.img -> boot/initrd.img-4.4.0-1-rpi
drwxr-xr-x 13 root root  4096 Mar 15 13:51 lib
drwx------  2 root root 16384 Mar 15 13:33 lost+found
drwxr-xr-x  2 root root  4096 Mar 15 13:40 media
drwxr-xr-x  2 root root  4096 Mar 15 13:40 mnt
drwxr-xr-x  3 root root  4096 Mar 15 13:51 opt
dr-xr-xr-x 86 root root     0 Jan  1  1970 proc
drwx------  2 root root  4096 Mar 15 13:40 root
drwxr-xr-x 14 root root   500 Mar 15 16:17 run
drwxr-xr-x  2 root root  4096 Mar 15 17:29 sbin
drwxr-xr-x  2 root root  4096 Mar 15 13:40 srv
dr-xr-xr-x 12 root root     0 Mar 15 18:14 sys
drwxrwxrwt  7 root root   140 Mar 15 16:17 tmp
drwxr-xr-x 10 root root  4096 Mar 15 13:40 usr
drwxr-xr-x 11 root root  4096 Mar 15 13:40 var
lrwxrwxrwx  1 root root    24 Mar 15 13:46 vmlinuz -> boot/vmlinuz-4.4.0-1-rpi

root@raspbian:~# ls -al /boot
total 59770
drwxr-xr-x  4 root root    16384 Jan  1  1970 .
drwxr-xr-x 21 root root     4096 Mar 15 13:46 ..
-rwxr-xr-x  1 root root     1494 Mar 15 13:51 LICENCE.broadcom
-rwxr-xr-x  1 root root  2115508 Mar 15 13:46 System.map-4.4.0-1-rpi
-rwxr-xr-x  1 root root    13328 Mar 15 13:51 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x  1 root root    13065 Mar 15 13:51 bcm2708-rpi-b.dtb
-rwxr-xr-x  1 root root    13101 Mar 15 13:51 bcm2708-rpi-cm.dtb
-rwxr-xr-x  1 root root    14411 Mar 15 13:51 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root    15081 Mar 15 13:51 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root    17932 Mar 15 13:51 bootcode.bin
-rwxr-xr-x  1 root root      156 Mar 15 13:51 cmdline.txt
drwxr-xr-x  4 root root     2048 Mar 15 13:33 config
-rwxr-xr-x  1 root root   132685 Mar 15 13:46 config-4.4.0-1-rpi
-rwxr-xr-x  1 root root      294 Mar 15 13:33 config-reinstall.txt
-rwxr-xr-x  1 root root       90 Mar 15 13:51 config.txt
-rwxr-xr-x  1 root root     6480 Mar 15 13:51 fixup.dat
-rwxr-xr-x  1 root root     2509 Mar 15 13:51 fixup_cd.dat
-rwxr-xr-x  1 root root     9713 Mar 15 13:51 fixup_db.dat
-rwxr-xr-x  1 root root     9713 Mar 15 13:51 fixup_x.dat
-rwxr-xr-x  1 root root  5877723 Mar 15 13:49 initrd.img-4.4.0-1-rpi
-rwxr-xr-x  1 root root     4009 Mar 15 13:33 installer-config.txt
-rwxr-xr-x  1 root root 11619747 Mar 15 13:33 installer-rpi1.cpio.gz
-rwxr-xr-x  1 root root 11629579 Mar 15 13:33 installer-rpi2.cpio.gz
-rwxr-xr-x  1 root root  4316272 Mar 15 13:33 kernel-rpi1_install.img
-rwxr-xr-x  1 root root  4514168 Mar 15 13:33 kernel-rpi2_install.img
-rwxr-xr-x  1 root root  4316272 Mar 15 13:33 kernel.img
drwxr-xr-x  2 root root     8192 Mar 15 13:51 overlays
-rwxr-xr-x  1 root root      140 Mar 15 13:33 post-install.txt
-rwxr-xr-x  1 root root      890 Mar 15 13:33 raspbian-ua-netinst-19700101T000020a.log
-rwxr-xr-x  1 root root  2743224 Mar 15 13:51 start.elf
-rwxr-xr-x  1 root root   615896 Mar 15 13:51 start_cd.elf
-rwxr-xr-x  1 root root  4934920 Mar 15 13:51 start_db.elf
-rwxr-xr-x  1 root root  3887400 Mar 15 13:51 start_x.elf
-rwxr-xr-x  1 root root  4316272 Mar 15 13:46 vmlinuz-4.4.0-1-rpi

root@raspbian:~# cat /boot/installer-config.txt
preset=minimal
packages=nano,ca-certificates,cron,less,python-minimal,hostapd,udhcpd
                          # comma separated list of extra packages
mirror=http://mirrordirector.raspbian.org/raspbian/
release=jessie
hostname=raspbian
boot_volume_label=        # Sets the volume name of the boot partition. The volume name can be up to 11 characters
                          # long. The label is used by most OSes (Windows, Mac OSX and Linux) to identify the
                          # SD-card on the desktop and can be useful when using multiple SD-cards.
domainname=raspbian.local
rootpw=raspbian
root_ssh_pubkey=          # public SSH key for root; on Debian "jessie" the SSH password login will be disabled for
                          # root if set; the public SSH key must be on a single line, enclosed in quotes
disable_root=             # set to 1 to disable root login (and password) altogether
username=                 # username of the user to create
userpw=                   # password to use for created user
user_ssh_pubkey=          # public SSH key for created user; the public SSH key must be on a single line, enclosed
                          # in quotes
user_is_admin=            # set to 1 to install sudo and make the user a sudo user
cdebootstrap_cmdline=
bootsize=+128M            # /boot partition size in megabytes, provide it in the form '+<number>M' (without quotes)
bootoffset=8192           # position in sectors where the boot partition should start. Valid values are > 2048.
                          # a bootoffset of 8192 is equal to 4MB and that should make for proper alignment
rootsize=                 # / partition size in megabytes, provide it in the form '+<number>M' (without quotes),
                          # leave empty to use all free space
timeserver=time.nist.gov
timezone=Etc/UTC          # set to desired timezone (e.g. Europe/Ljubljana)
locales=                  # a space delimited list of locales that will be generated during install
                          # (e.g. "en_US.UTF-8 nl_NL sl_SI.UTF-8")
system_default_locale=    # the default system locale to set (using the LANG environment variable)
disable_predictable_nin=1 # Disable Predictable Network Interface Names. Set to 0 if you want to use predictable
                          # network interface names, which means if you use the same SD card on a different
                          # RPi board, your network device might be named differently. This will result in the
                          # board having no network connectivity.
ifname=eth0
ip_addr=dhcp
ip_netmask=0.0.0.0
ip_broadcast=0.0.0.0
ip_gateway=0.0.0.0
ip_nameservers=
drivers_to_load=ax88179_178a
online_config=            # URL to extra config that will be executed after installer-config.txt
usbroot=                  # set to 1 to install to first USB disk
cmdline="dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline"
rootfstype=ext4
rootfs_mkfs_options=
rootfs_install_mount_options='noatime,data=writeback,nobarrier,noinit_itable'
rootfs_mount_options='errors=remount-ro,noatime'
final_action=reboot       # what to do at the end of install, one of poweroff / halt / reboot
hardware_versions=detect  # "detect" supports the install hardware only, set to "1 2" to produce an install that
                          # supports both Pi1 and Pi2
hwrng_support=1           # install support for the ARM hardware random number generator. The default is
                          # enabled (1) on all presets. Users requiring a `base` install are advised that
                          # `hwrng_support=0` must be added in `installer-config.txt` if HWRNG support is
                          # undesirable.
enable_watchdog=0         # loads up the hardware watchdog module and configures systemd to use it. Set to
                          # "1" to enable this functionality.
gpu_mem=32                # specify the amount of RAM in MB that should be reserved for the GPU
```

netsgnut, 2017-03-16:
It seems that for now the `raspberrypi-firmware-nokernel` package (`raspberrypi-bootloader-nokernel_1.20160523-1~nokernel2_armhf`) as used by Raspbian [does not contain the necessary dtb file](https://github.com/raspberrypi/firmware/commit/78c4983379de0537f49a97fb37ceaaf42632b28c). So until the Raspbian team upgrades the package, we can [use this fork which uses the Foundation's kernel and firmware package instead](https://github.com/FooDeas/raspberrypi-ua-netinst).

netsgnut, 2017-03-16:
BTW it is probably a similar issue to #446 despite the board difference. Also [link to the upstream bug](https://bugs.launchpad.net/raspbian/+bug/1658340). I will also try to [manually replace the files](https://github.com/debian-pi/raspbian-ua-netinst/issues/446#issuecomment-270525657) and see if it works as a dirty trick.

diederikdehaas, 2017-03-19:
> I suspect there may be a library/firmware missing

There may be files missing, but the main problem is that the firmware is just too old.
I submitted that upstream bug you linked to about it about 2 months ago, but still no response :-/
And tbh, I fear it won't get fixed ...

BenjaminHae, 2017-09-16:
Using the latest pre release (raspbian-ua-netinst v1.1.1-beta3) installation and booting afterwards is no problem.

Mausy5043, 2017-09-16:
@BenjaminHae : This is confusing. In #488 you're saying it's not working. Here you make it sound like it is. Which is it?

BenjaminHae, 2017-09-17:
In #488 the problem is wifi. Here it is booting.
Booting works, wifi does not.

Mausy5043, 2017-09-17:
Ah OK. Thanks

Mausy5043, 2017-09-17:
So this issue can now be closed?

BenjaminHae, 2017-09-17:
As far as I'm concerned yes.

Mausy5043, 2017-09-17:
@netsgnut : Please confirm that this issue is resolved when using the `v1.1.1-beta3` release.

netsgnut, 2017-09-18:
@Mausy5043 Sure, will verify on my side during the week here as well. Thanks.

Mausy5043, 2017-12-11:
@netsgnut : bump

netsgnut, 2017-12-12:
@Mausy5043 Sorry I have yet verified it here. Shall we close the issue for now first then? Thanks.

closed 2017-12-12 by Mausy5043