debian-pi/raspbian-ua-netinst #456: sudo: unable to resolve host (none)

CWempe, 2017-03-16:
I am using the v1.1.x branch.

In my post-install.txt I have the following commands.

```
chroot /rootfs /bin/cat /dev/zero | chroot /rootfs /usr/bin/ssh-keygen -t rsa -b 4096 -q -N ""
chroot /rootfs /bin/cat /dev/zero | chroot /rootfs /usr/bin/sudo -u user1 /usr/bin/ssh-keygen -t rsa -b 4096 -q -N ""
```

Both commands create the key files I want.
.
But the second one shows a warning.

    sudo: unable to resolve host (none)

And because of this issue, the comment in the key file is wrong ("(none)" instead of the real hostname).

How can I solve this issue?

Mausy5043, 2017-03-16:
I don't understand why you would want to pipe `/dev/zero` into `keygen`

Here's a snippet from my `post-install.txt`:
```
# Here we regenerate all the host SSH keys
rm /rootfs/etc/ssh/ssh_host_*
chroot /rootfs /usr/bin/ssh-keygen -N "" -t rsa   -b 4096 -f /etc/ssh/ssh_host_rsa_key
chroot /rootfs /usr/bin/ssh-keygen -N "" -t dsa   -b 1024 -f /etc/ssh/ssh_host_dsa_key
chroot /rootfs /usr/bin/ssh-keygen -N "" -t ecdsa -b 521  -f /etc/ssh/ssh_host_ecdsa_key
chroot /rootfs /usr/bin/ssh-keygen -N "" -t ed25519       -f /etc/ssh/ssh_host_ed25519_key
```

This works fine for me.  
Also not familiar with your second line. Could you explain what & why you're trying to accomplish there?

CWempe, 2017-03-16:
I added `/bin/cat /dev/zero`, because ssh-keygen asks about the passphrase.
This way I tell it to use no passphrase.
... in theory...

I will try again. :)

My second line creates a key for my personal user account "user1".

CWempe, 2017-03-17:
I still get the "none" hostname. 😞 

command in `post-install.txt`:

```
chroot /rootfs hostname
```

result:

```
(none)
```

Is this supposed to be this way?
After the installation `hostname` shows the correct value.

Everything else is working fine! 👍 
Thanks for your examples.

kpfleming, 2017-03-17:
The installer script populates /etc/hostname and /etc/hosts in /rootfs with the hostname you've configured (or 'pi' as a default), so no, it's not *supposed* to be this way. However, it appears that on Linux these files are not used by the library function called by the```hostname``` command, instead it calls the ```uname(2)``` function to get information from the kernel. Most likely ```ssh-keygen``` does the same thing. During the installation, the kernel has not been told what the machine's name is supposed to be.

To confirm this, add

```hostname $hostname```

in your ```post-install.txt``` file before the calls to ```ssh-keygen```. If this solves the problem, a simple patch to the installer script will fix this permanently.

CWempe, 2017-03-17:
I added your line `hostname $hostname`and it worked!

I would be happy to test this and make a pull request if you tell me where this line of code should be placed. 😄 

Between line 866 and 867?
https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L866

like:

```
# Tell the kernel what the hostname is supposed to be
hostname $hostname
```

kpfleming, 2017-03-18:
I would put it between 847 and 848, the block that is already doing system-level hostname configuration.

CWempe, 2017-03-18:
Ok. I created a PR.

https://github.com/debian-pi/raspbian-ua-netinst/pull/460

closed 2017-03-18 by CWempe