debian-pi/raspbian-ua-netinst #459: /boot/.git gets deleted during install

CWempe, 2017-03-18:
I started to organize my raspbian-ua-netinst configuration with git.

So I initiated a git repository in `/boot`.

But after my next installation, the `.git` folder and `.gitignore` are lost.
Looks these hidden files/folders do not get copied by line [223](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L223)

I will test a new build that copies these specific files.

CWempe, 2017-03-18:
I added this to `raspbian-ua-netinst/scripts/etc/init.d/rcS`.

```bash
# copy hidden git files from /boot if they exist
if [ -d "/boot/.git" ]; then
    echo "/boot/.git found. copying ..."
    cp -r -v -- /boot/.git /bootfs/ || fail
else.
    echo "/boot/.git not found."
fi
if [ -f "/boot/.gitignore" ]; then
    echo "/boot/.gitignore found. copying ..."
    cp -v -- /boot/.gitignore /bootfs/.gitignore || fail
else.
    echo "/boot/.gitignore not found."
fi

echo "/boot/"
ls -al /boot

echo "/bootfs/"
ls -al /bootfs
```

But the git files are still missing after the installation process.

Log shows this

```
Copying boot files... /boot/.git found. copying ...
/boot/.gitignore found. copying ...
/boot/
total 59962
drwxr-xr-x    6 0        0            16384 Jan  1 00:00 .
drwxrwxr-x   17 500      1000           360 Jan  1 00:00 ..
drwxr-xr-x    8 0        0             2048 Mar 18  2017 .git
-rwxr-xr-x    1 0        0              259 Mar 18  2017 .gitignore
-rwxr-xr-x    1 0        0             1494 Mar 18  2017 LICENCE.broadcom
-rwxr-xr-x    1 0        0                0 Mar 18  2017 README.md
drwxr-xr-x    2 0        0             2048 Mar 18  2017 System Volume Information
-rwxr-xr-x    1 0        0          2177376 Mar 18  2017 System.map-4.4.0-1-rpi2
-rwxr-xr-x    1 0        0            13328 Mar 18  2017 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x    1 0        0            13065 Mar 18  2017 bcm2708-rpi-b.dtb
-rwxr-xr-x    1 0        0            13101 Mar 18  2017 bcm2708-rpi-cm.dtb
-rwxr-xr-x    1 0        0            14411 Mar 18  2017 bcm2709-rpi-2-b.dtb
-rwxr-xr-x    1 0        0            15081 Mar 18  2017 bcm2710-rpi-3-b.dtb
-rwxr-xr-x    1 0        0            17932 Mar 18  2017 bootcode.bin
-rwxr-xr-x    1 0        0               73 Mar 18  2017 cmdline.txt
drwxr-xr-x    4 0        0             2048 Mar 18  2017 config
-rwxr-xr-x    1 0        0           132692 Mar 18  2017 config-4.4.0-1-rpi2
-rwxr-xr-x    1 0        0              294 Mar 18  2017 config-reinstall.txt
-rwxr-xr-x    1 0        0              294 Mar 18  2017 config.txt
-rwxr-xr-x    1 0        0             6480 Mar 18  2017 fixup.dat
-rwxr-xr-x    1 0        0             2509 Mar 18  2017 fixup_cd.dat
-rwxr-xr-x    1 0        0             9713 Mar 18  2017 fixup_db.dat
-rwxr-xr-x    1 0        0             9713 Mar 18  2017 fixup_x.dat
-rwxr-xr-x    1 0        0          5645099 Mar 18  2017 initrd.img-4.4.0-1-rpi2
-rwxr-xr-x    1 0        0              445 Mar 18  2017 installer-config.txt
-rwxr-xr-x    1 0        0         11602853 Mar 18  2017 installer-rpi1.cpio.gz
-rwxr-xr-x    1 0        0         11611936 Mar 18  2017 installer-rpi2.cpio.gz
-rwxr-xr-x    1 0        0          4316272 Mar 18  2017 kernel-rpi1_install.img
-rwxr-xr-x    1 0        0          4514168 Mar 18  2017 kernel-rpi2_install.img
-rwxr-xr-x    1 0        0          4514168 Mar 18  2017 kernel7.img
drwxr-xr-x    2 0        0             8192 Mar 18  2017 overlays
-rwxr-xr-x    1 0        0             4447 Mar 18  2017 post-install.txt
-rwxr-xr-x    1 0        0          2743224 Mar 18  2017 start.elf
-rwxr-xr-x    1 0        0           615896 Mar 18  2017 start_cd.elf
-rwxr-xr-x    1 0        0          4934920 Mar 18  2017 start_db.elf
-rwxr-xr-x    1 0        0          3887400 Mar 18  2017 start_x.elf
-rwxr-xr-x    1 0        0          4514168 Mar 18  2017 vmlinuz-4.4.0-1-rpi2
[... 19 lines cut ...]
-rwxr-xr-x    1 0        0              294 Jan  1 00:00 config-reinstall.txt
-rwxr-xr-x    1 0        0              294 Jan  1 00:00 config.txt
-rwxr-xr-x    1 0        0             6480 Jan  1 00:00 fixup.dat
-rwxr-xr-x    1 0        0             2509 Jan  1 00:00 fixup_cd.dat
-rwxr-xr-x    1 0        0             9713 Jan  1 00:00 fixup_db.dat
-rwxr-xr-x    1 0        0             9713 Jan  1 00:00 fixup_x.dat
-rwxr-xr-x    1 0        0          5645099 Jan  1 00:00 initrd.img-4.4.0-1-rpi2
-rwxr-xr-x    1 0        0              445 Jan  1 00:00 installer-config.txt
-rwxr-xr-x    1 0        0         11602853 Jan  1 00:00 installer-rpi1.cpio.gz
-rwxr-xr-x    1 0        0         11611936 Jan  1 00:00 installer-rpi2.cpio.gz
-rwxr-xr-x    1 0        0          4316272 Jan  1 00:00 kernel-rpi1_install.img
-rwxr-xr-x    1 0        0          4514168 Jan  1 00:00 kernel-rpi2_install.img
-rwxr-xr-x    1 0        0          4514168 Jan  1 00:00 kernel7.img
drwxr-xr-x    2 0        0             1580 Jan  1 00:00 overlays
-rwxr-xr-x    1 0        0             4447 Jan  1 00:00 post-install.txt
-rwxr-xr-x    1 0        0          2743224 Jan  1 00:00 start.elf
-rwxr-xr-x    1 0        0           615896 Jan  1 00:00 start_cd.elf
-rwxr-xr-x    1 0        0          4934920 Jan  1 00:00 start_db.elf
-rwxr-xr-x    1 0        0          3887400 Jan  1 00:00 start_x.elf
-rwxr-xr-x    1 0        0          4514168 Jan  1 00:00 vmlinuz-4.4.0-1-rpi2
```

Am I missing something?
Is there a cleaning process that deleted some files afterwards?
Or is there another boot-partition that the files hould be copied to?

Mausy5043, 2017-03-19:
I don't understand why you are doing this. The installer itself is automatically generated. Why would you want to make a git repo out of it?

Hint: check the code around line [L896](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L896).

CWempe, 2017-03-19:
I just want to organize my custom files in a repo.
To back them up and to keep track of the changes I made.

Kind of like other people keep their entire /etc files in a repo.
see: https://help.ubuntu.com/lts/serverguide/etckeeper.html

This way I can edit some custom files in /boot, commit, push and reboot to test the new configuration.
And whenever I want to setup a new Pi, I just extract the latest release to the sdcard and `git clone/fetch` my config to it.

Thanks for your hint.
I did not kknow I had to copy the files another time (old-boot -> bootfs -> new-boot).

I will create a pull request and we can discuss weather this is usefull feature to the project or not. 😄

Mausy5043, 2017-03-19:
you might want to take a look at [this repo](https://github.com/Mausy5043/mod-raspbian-ua-netinst). I think it does what you need.

goranche, 2017-03-19:
a git repository on the boot partition shouldn't be copied over to the target system...

it's something that useful only for a very specific use case, and can easily be done in the post-install script, so I wouldn't add it to the installer.

CWempe, 2017-03-19:
@Mausy5043 
You are right, that repo sounds like my use case.
But I would like to stick to the original repo. :)

My next approach would be to take a step back and ask it the hidden files are left out on purpose.
Changing
`cp -r -- /bootfs/* /boot`
to
`cp -r -- /bootfs/. /boot`
would copy all files (including hidden).
This would not only copy my git files, but every file (I think).
Now, I am wondering why `/boot/.gitignore` is not copied, but `/boot/config/files/root/*.bashrc` is. 😕 

Anyway, if you two (as the experts) are saying this should not be implemented, I respect that. 😃 

I will try to solve this via the post-install script, like @goranche suggested.

Thanks. 👍

Mausy5043, 2017-03-19:
The repo I pointed you is a separate repo that **modifies**  `raspbian-ua-netinst`. It is not a branch of `raspbian-ua-netinst`.

Mausy5043, 2017-03-19:
> Now, I am wondering why /boot/.gitignore is not copied, but /boot/config/files/root/*.bashrc is.

`cp` does not copy dot-files like `.gitignore` because the dot at the start means it is a hidden file. It is also hidden to `cp` unless the shell or `cp` are instructed otherwise.
 The files in the `.../config/files/root/` path are specifically mentioned in a separate file. Each entry in the file is fed to `cp` in turn. So, in that case `cp` is specifically told about the existence of the file and therefore it can copy it.

closed 2017-03-22 by Mausy5043