debian-pi/raspbian-ua-netinst #461: send status mail after installation finished

CWempe, 2017-03-18:
Would it make sense to integrate an option to inform the user about the completion of the installation?

If you run your installation without a monitor you don't know exactly when the installation is finished.

We could integrate something like this:

```
finished_mail=user@example.com
```

At the end of the installation script there could be something like this:

```
if (finished_mail is set)
then
   send mail to user
else
   dont
```

Of cause the user has to make sure the system can send mails.

It could be combined optionally with #338.

Mayby I would try to integrate this feature if it fits the project.
Otherwise I would do this in the `post‑install.txt`. :)

Mausy5043, 2017-03-19:
I've implemented this functionality as part of my own post-first-boot activities. I doubt if it would be something worthy of making the installer.

goranche, 2017-03-19:
doesn't really make sense for an installer like this...
way too many options, and can easily be done in post-install

closed 2017-03-19 by goranche