debian-pi/raspbian-ua-netinst #468: bug in RASPBIAN  instalator 2017-04-10

GanFall, 2017-05-23:
Hi can someone help me to check this issue? I have tried to modify raspberry installation. I would like to use 
latest RASPBIAN as it is, but only make root fs smaller then full flashcard size. I have 16GB flashcard I and want have os partition only size of 12GB. It is my security measurement, because every SD card has different size and when sdcard fails, it is hard to move backup img to another card. I know procedure of change size of FS after OS has been installed, but it take to much of time and feels not clear
then original installation without bustling with fs.
I tried to use **installer-config.txt** file, but it simply doesn't work. What ever I try. I have put it on every partition of SD card now. Event to /root/ folder  . Documentation is not clear for me about that. But now I feel that this must be some kind of bug. Please help :o)

Version:April 2017
Release date:2017-04-10
Kernel version:4.4

goranche, 2017-05-23:
this is something that has been extensively tested (and used by myself on an almost weekly basis), so I highly doubt it's a bug...

if you set the `bootsize` and/or `rootsize` variables as described in the [README.md - Installer customization](https://github.com/debian-pi/raspbian-ua-netinst#installer-customization), the partition sizes will be created accordingly.

Also, please take a look at the [reporting issues section in the CONTRIBUTING.md](https://github.com/debian-pi/raspbian-ua-netinst/blob/master/CONTRIBUTING.md#reporting-issues) file, and provide the data listed there.

but basically: you'll have to add the two mentioned settings to `installer-config.txt` before booting up the installer, and that's pretty much it

GanFall, 2017-05-25:
Hi,I did it exactly as documented. I'm only not fully sure about location of that file.  Can you please confirm where are should put this file? Which one location is really correct. I already tested it maybe 5 times. No luck with any location. I have RPi2 hopefully this function is not limited to other version. I have tried to use NOOBS or directly raspbian image also not working.  I want only to adjust root size. So I have added only "rootsize=+8000M" maybe this is the problem. I have worries that SDcard will be not able handle so many installations :o) Br, Milos

goranche, 2017-05-25:
from the link to the README.md I've provided in my previous comment:
`When you've written the installer to a SD card, you'll see a file named cmdline.txt and you create the installer-config.txt file alongside that file.`

So, this is when you write the installer to the SD card, not after the installer goes through the installation process.

GanFall, 2017-05-25:
Hi Thanks, I just hoped that there something else what I have missed. Because there are lot of resources on internet. Even that is not working. After first boot, the FS is expanded over entire SDcard.

goranche, 2017-05-25:
I'm sorry, but without additional information (as stated in my first comment) there isn't much more I can do for you.

GanFall, 2017-05-25:
I will give it one more shot, need to update anyway. If it will after first boot expanded again over entire sdcard I will try to report it as bug. I would like to provide more details, but there is no much what to check. I have created file installer-config.txt with that rootsize=+8000M variable placed it alongside with cmdline.txt  and make first boot in rpi. After I have found that the FS is again over entire SDcard I have made format and tried to add that file on every possible place after mounted in linux on newly installed card.  The situation has repeated and the FS was again same size as sdcard after first boot in rpi.  I thank you, now I at least know that nothing is missing :o)

goranche, 2017-05-25:
I'm closing this issue  for now, since I'm pretty certain that this isn't a bug.
It has been tested many times before; most of my custom installers use the functionality, and it works, I'm certain someone else would have reported it as well.

Should you feel I've closed this prematurely, you're free to reopen it, of course, but do provide more information in that case. 😉

closed 2017-05-25 by goranche