debian-pi/raspbian-ua-netinst #472: Firmware Package Problem

Zeerix, 2017-06-04:
Good news and bad news.

Good news: They updated the package raspberrypi-bootloader-nokernel from version 1.20160523-1 to 1.20170427-1.
Bad news: The installer doesn't boot anymore [1]. My guess is that the kernel is too old for the new firmware.

[1] That is, a newly build installer. I am currently testing, if an older installer creates a working system.

Zeerix, 2017-06-05:
Sorry for the panic. I just tried with my RPi 2 and all is well.
After some testing, I found that "enable_uart=1" seems to be the problem. With it, the kernel panics (on RPi 3), without it, the device boots.

goranche, 2017-06-05:
https://github.com/raspberrypi/firmware/issues/798

can this be closed?

goranche, 2017-06-05:
ooops... closed it  😊   my bad, didn't intend to

if you want, reopen it, @Zeerix

Zeerix, 2017-06-05:
@goranche I don't have a button to reopen. Probably no permissions to do so.

And why would you close it? It's not resolved, even if this only applies to RPi3 now.

goranche, 2017-06-05:
because it's a known issue in the firmware (click link), and nothing we can do anything about until the new firmware with the fix is released...

Zeerix, 2017-06-05:
We can remove "enable_uart=1".

goranche, 2017-06-05:
👍  so we have a workaround... and the issue is resolved for now...

CWempe, 2017-07-10:
After facing the same issue, I would suggest to mention this problem and the workaround in the release notes.
It is really frustrating if you try the latest image and get a kernel panic.

More, I would suggest to build a new image with "enable_uart=1" and kernel 4.9 to fix #471 as well.
Name it beta2, maybe.

But as I can build my own image, its just a suggestion to make it easier for the average user. :)

Mausy5043, 2017-12-09:
Closing this issue for now, assuming it is resolved. If you feel the closure is in error, please feel free to re-open and add new information.

closed 2017-12-09 by Mausy5043