debian-pi/raspbian-ua-netinst #474: 2 Issues before installation process | seems to be network related?

msaptrta, 2017-06-06:
Hello,
After following the installation instructions and booting the pi, I ran into a few errors.
    The first set of errors I got was a complete set of "Failed to set time using ntpdate!  Failed to set time using timeserver 'time.nist.gov.'.  Failed to set time using timeserver 'time.nist.gov.'.  Failed to set time using timeserver 'nist1.symmetricom.com'.  Failed to set time ..." etc.  
    At the end of all these failed time sets, the console stated, "Make sure that rdate port 37 is not blocked by your firewall."  So that's what I did.  It was alright so I tried again, yet same errors occurred.  Any thoughts or ideas how to fix this?

    The next error I got during the "Starting install process...". The console stated, "P: Retrieving Release \n wget: bad address 'mirrordirector.raspbian.org' \n E: Couldn't download Release! \n\n Oh noes, something went wrong! \n You have 10 seconds to hit ENTER to get a shell..."

    Having both of these errors leads me to believe it's a network related issue, but I'm not sure.  Any help or suggestions would be appreciated.

Mausy5043, 2017-06-06:
Thank you for trying the raspbian-ua-netinst installer. This does indeed sound like a network related problem.

The installer creates a log file, which is probably located in `/boot/`. Please provide the log.

msaptrta, 2017-06-07:
Well I was able to get the time sync to work by connecting to a different network.  It may be an access 
rule on the other network that was causing the first issue; however, the second problem still persist.
I also checked in the /boot/ directory but there was no log that had been created.

diederikdehaas, 2017-06-08:
Sounds like a DNS issue.
When you get the 
```
Oh noes, something went wrong! 
 You have 10 seconds to hit ENTER to get a shell...
```
do hit ENTER and you'll get a minimal/busybox shell. Check whether `/etc/resolv.conf` exists and that it contains the proper values for your network. Trying a `ping google.com` would also give you a strong indication whether DNS is working or not

Mausy5043, 2017-06-11:
When you open the SD card on another computer you should see a file called  `raspbian-ua-netinst-<date>.log`  where `<date>` is the date/time of the failed install.

Mausy5043, 2017-06-11:
There's definitely an upstream problem. This morning's install:
```
=================================================
raspbian-ua-netinst
=================================================
Revision git~93d2a91
Built on Sun Jun 11 10:09:10 CEST 2017
Running on Raspberry Pi version unknown
=================================================
https://github.com/debian-pi/raspbian-ua-netinst/
=================================================
Starting HWRNG succeeded!
Copying boot files... OK
=================================================
=== Start executing installer-config.txt. ===
=== Finished executing installer-config.txt. ===
=================================================

Network configuration:
 ip4_addr = dhcp
 ip6_addr = disable
 online_config = 

Waiting for eth0... 
OK
Configuring eth0 for IPv4 using DHCP... 10.0.1.225/24
Time set using ntpdate to... Sun Jun 11 08:16:26 UTC 2017

Installation started at Sun Jun 11 08:16:07 UTC 2017.

=================================================
    No Raspberry Pi hardware detected!!
Value of cpu_revision variable: '0000'
         Building for Pi 1, 2 and 3.         
=================================================
```

Mausy5043, 2017-06-11:
End of the log:
```
Configuring apt:
  Copying raspbian repo to sources.list... OK
  Replacing __RELEASE__ with jessie... OK
  Adding raspberrypi.org gpg key to apt-key.
  Copying ./raspberrypi.org.list to /etc/apt/sources.list.d/... OK
  Copying ./01vcgencmd.pref to /etc/apt/preferences.d/... OK
Updating package lists... OK
Installing bootloader package (=firmware)... OK
Installing libraspberrypi-bin package (=vcgencmd,raspistill,raspivid,etc)... OK
Configuring bootloader to start the installed system...
Adding boot config for kernel version: '4.9.0-2-rpi'
Copying kernel to /boot/kernel.img... OK
Adding boot config for kernel version: '4.9.0-2-rpi2'
Copying kernel to /boot/kernel7.img... OK
Adding boot config for kernel version: '4.9.0-2-rpi2'
Creating default cmdline.txt... OK
=================================================
=== Start executing post-install.txt. ===
```

After the reboot:
```
pi@rbian ~ $ uname -a
Linux rbian 4.9.0-2-rpi #1 Raspbian 4.9.13-1+rpi3 (2017-05-18) armv6l GNU/Linux
pi@rbian ~ $ cat /proc/cpu
cpu/     cpuinfo
pi@rbian ~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 0010
Serial		: 00000000b72f8406
```

Odd.

Mausy5043, 2017-06-11:
If you have MAC-address filtering on your firewall, you may be experiencing problems related to  #471.

Mausy5043, 2017-10-17:
No additional info after 4 months.
Assuming the problem went away. Closing this issue now. 
If you feel this issue was closed in error and you want to add additional info, please feel to re-open.

Mausy5043, 2017-12-09:
Closing this issue for now, assuming it is resolved. If you feel the closure is in error, please feel free to re-open and add new information.

closed 2017-12-09 by Mausy5043