debian-pi/raspbian-ua-netinst #476: Kernel Panic on first boot of branch v1.1.1-beta (ed24416)

natarajmb, 2017-07-10:
Raspberry Pi 2 B+ 
64GB SDCard

Steps to create:
1. Format the SD card on macOS with MS-DOS option
2. Downloaded zip release v1.1.1-beta (ed24416)
3. Copied the content to formatted SD Card (no installer-config.txt i.e. all default options)
4. Powered on RPI2 with above SD Card

Result:
Kernel panic - not syncing : VFS: Unable to mount root fs on unknown-block(0,0)

I have also tried flashing .img release directly to SD Card with same result

[image]

CWempe, 2017-07-14:
see: https://github.com/debian-pi/raspbian-ua-netinst/issues/472

edit `config.txt`and set `enable_uart=0`

natarajmb, 2017-07-14:
I tried the above suggestion but result is same. FYI, same hardware works with stable release.

diederikdehaas, 2017-07-19:
The new kernel got merged in recently, so if you can build it yourself and verify, that would be great.
I tested that PR (https://github.com/debian-pi/raspbian-ua-netinst/pull/475) with a RPi 2B+ and it worked fine :)

Mausy5043, 2017-07-19:
@natarajmb You might also want to consider throwing in #480 in your test

natarajmb, 2017-07-20:
@diederikdehaas and @Mausy5043 I have tested both the commits independently and it works. Thanks for all your help.

Mausy5043, 2017-07-25:
Please check if the beta2 release gives better results.

natarajmb, 2017-07-26:
Sure will do. With beta1, I was having issue with driver loading for a wifi dongle. I will report if this release fixes it. FYI this is the option i'm using for driver loading (hardware: edimax ew-7811un) in installer-config.txt 

drivers_to_load=8192cu

Error I'm seeing is this, I'm assuming I haven't build the kernel properly so will try beta2 and report back

> unknown symbol in module or unknown parameter 8192cu

Mausy5043, 2017-07-26:
You also have `ifname=wlan0` set?
`wpa_supplicant.conf` supplied?

natarajmb, 2017-07-26:
Yes, I do have `ifname` in installer-config.txt and wap_supplicant.conf copied through `files` folder at right location. Attaching all three files for brevity. 

`wpa_supplicant.conf` -> <ROOT>/config/files/etc/wpa_supplicant/
`interfaces`-> <ROOT>/config/files/etc/network/

[wpa_supplicant.conf.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/1177368/wpa_supplicant.conf.txt)
[interfaces.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/1177369/interfaces.txt)

I haven't tried yet with latest beta and will let you know if it works!

Zeerix, 2017-07-26:
> unknown symbol in module or unknown parameter 8192cu

The driver needs `cfg80211.ko` with current kernel, which was not added to the installer. #482 fixes that.

Also with #469 merged, you don't need to specify `drivers_to_load` anymore.

natarajmb, 2017-07-26:
Raspberry Pi 2 B+
Tested with beta2, unfortunately i'm hitting same issue as the first post i.e. kernel panic with default setting no customisations.

[image]

natarajmb, 2017-07-26:
Just tested building on my own from commit 3daacb3 and it works. I think last commit #469 has broken it.

Zeerix, 2017-07-26:
> Just tested building on my own from commit 3daacb3 (#482) and it works. I think last commit #469 has broken it.

I disagree. I tried `beta2` which goes into kernel panic, and then build the last commit (`caace15`) myself which worked.

The problem is that the release ZIP files (and img.* as well) for both beta1 and beta2 contain broken installer-cpio files, which are only 1 KB instead of a couple of MB.

natarajmb, 2017-07-27:
Good point, let me build on my own with last commit and see if it works. I only took the beta2 release and when it didn't work went one commit behind.

natarajmb, 2017-07-27:
I jumped the gun earlier, its working properly with latest commit `3daacb3`. @Zeerix as you said its the release that has the wrong installer-cpio.

Zeerix, 2017-08-02:
@Mausy5043 Should I open a separate issue about the problem with the release files?

Mausy5043, 2017-08-03:
yes. please do. I'm low on time to look into this right now. Can't understand why the `cpio` file doesn't get

Mausy5043, 2017-08-03:
If someone can confirm that building from source (current state of `v1.1.x`) works. That would be nice.

Mausy5043, 2017-08-04:
There's now also a beta3 version. I am pretty sure the `cpio` files are not corrupted in that one.

Mausy5043, 2017-08-10:
Closing since the beta3 seems to fix this issue.
If you believe the closure was premature, please re-open and add new information.

closed 2017-08-10 by Mausy5043