debian-pi/raspbian-ua-netinst #478: Buggy on Galaxy S5

mibzman, 2017-07-15:
### What did you try to accomplish?
use the wifi network on my Galaxy S5
### What steps did you take in order to accomplish that?
run the script as directed on the readme on a fresh install of raspian
### What did you expect to happen?
the captive portal to work on all devices
### What actually happened?
the captive portal works on all my devices except my galaxy s5.  the portal works correctly on my mac, other android phone and ipad without a problem.

Mausy5043, 2017-07-19:
Uploading a copy of the installer log would be helpful
Which version of the installer are you using?

mibzman, 2017-08-01:
Unfortunatly I no longer have the device so I can't upload the logs, but I believed i used the https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.0.9 release.

Based on what the newer android devices did when they tried to log into a captive portal, my older android device would navigate to whatever address it was given, while the newer ones would navigate to 'google.com' and expect to be redirected.

kpfleming, 2017-08-01:
raspbian-ua-netinst is an installer for raspbian; it's not responsible for packages which you install from the Raspbian repository. There's certainly no 'captive portal' package installed by default with this installer. If you've installed additional packages from the Raspbian repository, you'll either need to use a general Raspbian support forum, or you'll need to use support resources from the projects who produced those packages.

Mausy5043, 2017-10-31:
No additional comments/info after 3 months. Closing this issue. If you feel the issue was closed in error, please feel free to re-open and add additional info.

closed 2017-10-31 by Mausy5043