debian-pi/raspbian-ua-netinst #484: dpkg issues

CustomIRCd, 2017-07-28:
Whilst watching the installation via a TV (yes i'm that sad) i noticed loads of errors with dpkg pre-dependency problem

Running on RPi3 with latest release (1.1.x) from github. I built the image on a server running ubuntu (latest)

Here's the log
https://gist.github.com/CustomIRCd/8305dcbfcc2442ad88425f217e86ee4c

is this something i could solve with a simple command?

BenjaminHae, 2017-07-28:
Excerpt from Readme:
Note: During the installation you'll see various warning messages, like "Warning: cannot read table of mounted file systems" and "dpkg: warning: ignoring pre-dependency problem!". Those are expected and harmless.

CustomIRCd, 2017-07-28:
Thanks @BenjaminHae  - i'm not normally that sad to sit and watch it. I'll close this issue then

closed 2017-07-28 by CustomIRCd