debian-pi/raspbian-ua-netinst #487: mounting rootfs failed - v1.1.1-beta3 release

johr, 2017-08-19:
tried to install type "server" with raspbian-ua-netinst v1.1.1-beta3 release.
installer stopped when trying to mount the rootfs with:
"Mounting new filesystems... mount: mounting /dev/mmcblk0p2 on /rootfs failed: Invalid argument"

I did not connect a screen to check it directly on the raspi (1,3) but tested it on my desktop, seems that data=writeback was the reason for the problem. 
At least when I set the `rootfs_install_mount_options=noatime,nobarrier,noinit_itable` the error did not happen! 

details:

installer-config.txt:
```
preset=server
packages=screen,zip,unzip,vim
release=jessie
hostname=baseserver
domainname=baseserver.local
disable_root=1
username=admin
userpw=admin
user_is_admin=1
rootsize=+1024 
timeserver=time.nist.gov
timezone=Europe/Vienna
locales=de_AT.utf8
system_default_locale=de_AT.utf8
#apt proxy
http_proxy=http://10.0.0.88:3142
export http_proxy
```
and  the log-file:

```
=================================================
raspbian-ua-netinst
=================================================
Revision git~caace15
Built on Thu Aug  3 17:49:32 CEST 2017
Running on Raspberry Pi version 3
=================================================
https://github.com/debian-pi/raspbian-ua-netinst/
=================================================
Starting HWRNG succeeded!
Copying boot files... OK
=================================================
=== Start executing installer-config.txt. ===
=== Finished executing installer-config.txt. ===
=================================================

.....

Installer configuration:
  preset = server
  packages = screen,zip,unzip,vim,sudo
  mirror = http://mirrordirector.raspbian.org/raspbian/
  release = jessie
  hostname = baseserver
  domainname = baseserver.local
  rootpw = raspbian
  root_ssh_pubkey = 
  disable_root = 1
  username = admin
  userpw = admin
  user_ssh_pubkey = 
  user_is_admin = 1
  cdebootstrap_cmdline = --flavour=minimal --include=linux-image-rpi2-rpfv,cpufrequtils,kmod,libpam-systemd,rng-tools,fake-hwclock,ifupdown,net-tools,ntp,openssh-server,dosfstools,resolvconf,vim-tiny,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,less,man-db,isc-dhcp-client,screen,zip,unzip,vim,sudo
  boot_volume_label = 
  bootsize = +128M
  bootoffset = 8192
  rootsize = +1024
  timeserver = time.nist.gov
  timezone = Europe/Vienna
  locales = de_AT.utf8
  system_default_locale = de_AT.utf8
  cmdline = dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline
  usbroot = 
  rootdev = /dev/mmcblk0
  rootpartition = /dev/mmcblk0p2
  rootfstype = ext4
  rootfs_mkfs_options = 
  rootfs_install_mount_options = noatime,data=writeback,nobarrier,noinit_itable
  rootfs_mount_options = errors=remount-ro,noatime
  final_action = reboot
  hardware_versions = 3
  gpu_mem = 

Waiting 5 seconds.....
Waiting for /dev/mmcblk0... OK
Applying new partition table... OK
Initializing /boot as vfat... OK
Copying /boot files in... OK
Initializing / as ext4... OK
Mounting new filesystems... mount: mounting /dev/mmcblk0p2 on /rootfs failed: Invalid argument

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```
putting 

```
rootfs_install_mount_options=noatime,nobarrier,noinit_itable   
```   
without data=writeback worked.

Mausy5043, 2017-08-20:
Hmmm. I missed this, when testing because I always use `f2fs` on SD-cards.

Mausy5043, 2017-08-20:
Further, 'data=writeback' is a valid [mount option for ext4](https://www.kernel.org/doc/Documentation/filesystems/ext4.txt). So, it is unlikely that it is the cause of your problem. More like a symptom.
Have you tried using another SD-card? The current one may have expired.

Mausy5043, 2017-08-20:
And also I see you're setting the parameter `rootsize=+1024`.
You might want to try leaving it empty or setting that as `rootsize=+1024M`.

johr, 2017-08-20:
just used the defaults which is ext4 ...
it is a valid option but only if the filesystem supports journaling. maybe the filesystem was not created with has_journal ... but thats no expert knowledge from my side, but just what i found out searching. so it may be more of a symptom ...

tested with 2 different sd-cards and raspis.
 
will try your proposed change for rootsize ... maybe some strange side-effect.

but using not writeback should only be relevant for the installer not the installed system, cause the  installed system will be mount with other options, right?

Mausy5043, 2017-08-20:
>  cause the installed system will be mount with other options, right?

Yes, those are `errors=remount-ro,noatime`

johr, 2017-08-20:
changed:
rootsize=+1024M 
exactly that way with appended 'M' (seems the other form was from some very old installer-config ...)
it worked without special handling of mount-options resp. mkfs options!

many thanks for the hint.

closed 2017-08-20 by johr