debian-pi/raspbian-ua-netinst #488: Wifi install on Pi Zero W

BenjaminHae, 2017-09-10:
I'm trying to get wifi to work on a pi zero w during installation. When using raspbian lite everything works as expected, so I've included all kernel modules from that in the installer:

     ifname=wlan0
     drivers_to_load="brcmfmac brcmutil cfg80211 rfkill"

But when booting the installer it aborts after `Waiting for wlan0`:

	[...]
	Network configuration:
	  ip4_addr = dhcp
	  ip6_addr = disable
	  online_config = 
	
	Loading drivers.
	Finished loading drivers.
	Loading additional drivers.
	 Loading driver 'brcmfmac' ... OK
	 Loading driver 'brcmutil' ... OK
	 Loading driver 'cfg80211' ... OK
	 Loading driver 'rfkill' ... OK
	Finished loading additional drivers
	Waiting for wlan0... 
	1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. FAILED
	
	Oh noes, something went wrong!
	You have 10 seconds to hit ENTER to get a shell...

What steps can I take to gather more information on this issue? Or is this depending on #455 ?

BenjaminHae, 2017-09-16:
After installing the system using a LAN adapter wifi still doesn't work. The interface is not even visible when executing `ifconfig` or `ip link show`.
I tried `rpi-update` to verify that it's not because of the old firmware, but that didn't help either. As far as I can see the relevant kernel modules are being loaded.

Zeerix, 2017-09-17:
If you install using LAN (i.e. ifname=eth0), the installed system does not have Wifi configured.
Only the network interface you are using for installation will be configured in the installed system.

BenjaminHae, 2017-09-17:
I would still expect the interface to be available.
Even when I configure `wlan0` in `/etc/network/interfaces` and call `ifup wlan0` the output is
```
Cannot find device "wlan0".
Failed to bring up wlan0.
```
What other steps are necessary to configure the device?

Mausy5043, 2017-09-17:
Could you post the output of `lsmod`?

BenjaminHae, 2017-09-17:
```
root@artnet_node:~# lsmod
Module                  Size  Used by
brcmfmac              267114  0 
brcmutil                9911  1 brcmfmac
cdc_ether               6246  0 
r8152                  41022  0 
cfg80211              525998  1 brcmfmac
rfkill                 21373  1 cfg80211
bcm2835_gpiomem         3791  0 
uio_pdrv_genirq         3718  0 
fixed                   3029  0 
uio                    10166  1 uio_pdrv_genirq
fuse                   94600  3 
ipv6                  384391  20
```
As far as I found out `brcmfmac` is the one implementing the driver for the wifi of the zero w.

Mausy5043, 2017-09-17:
Please note that we don't have access to a PZW.

Mausy5043, 2017-09-17:
Please also post your `installer-config.txt` and `post-install.txt`.

BenjaminHae, 2017-09-17:
`installer-config.txt`:
```
hostname=artnet_node
packages=unionfs-fuse
ifname=eth0
timezone=Europe/Berlin
```

`post-install.txt`
```
################################################
### Make the system read-only

rm /rootfs/etc/mtab
ln -s /proc/self/mounts /rootfs/etc/mtab

# Package should've been installed by installer-config.txt, but still...
chroot /rootfs apt-get -y install unionfs-fuse
# install olad
chroot /rootfs apt-get -y install ola
# install access point stuff
chroot /rootfs apt-get -y install dnsmasq hostapd

mkdir -p -m 700 /rootfs/root/.ssh
cp /bootfs/config/authorized_keys /rootfs/root/.ssh/authorized_keys
chmod 600 /rootfs/root/.ssh/authorized_keys

# configure access point
cp /bootfs/config/hostapd.conf /rootfs/etc/
cp /bootfs/config/dnsmasq.conf /rootfs/etc/
sed -i 's#\#DAEMON_CONF.*$#DAEMON_CONF="/etc/hostapd/hostapd.conf"#' /rootfs/etc/default/hostapd

cp /bootfs/mount_unionfs /rootfs/usr/local/bin/mount_unionfs
chmod +x /rootfs/usr/local/bin/mount_unionfs

sed -i 's/^\/dev\/mmcblk\S*\s\+\/[a-z]*\s\+\S*\s\+/&ro,/' /rootfs/etc/fstab
echo "mount_unionfs /var fuse defaults 0 0" >> /rootfs/etc/fstab

# activate ipv6
sed -i "/wlan0 inet dhcp/a\\
iface wlan0 inet6 dhcp" /rootfs/etc/network/interfaces

mv /rootfs/var /rootfs/var_orig
mkdir /rootfs/var /rootfs/var_rw
mount --bind /rootfs/var_orig /rootfs/var

```
The lines after `#activate ipv6` don't have any effect as the line `wlan0 inet dhcp` is not available.

Mausy5043, 2017-09-17:
I'd expect using 
```
 ifname=wlan0
 drivers_to_load="brcmfmac brcmutil cfg80211 rfkill"
```
should let you boot with WiFi enabled.
Make sure you have `/boot/config/wpa_supplicant.conf` available though or you won't get a connection 😉 .

And if you could try that with the `v1.1.1-beta3` release, I'm curious what the output would be.

BenjaminHae, 2017-09-17:
That's exactly what I tried at first :) (for output see first post). Exactly this configuration with `v1.1.1-beta3`. `wpa_supplicant.conf`:
```
network={
    ssid="YYYYY"
    psk="XXXXX"
    key_mgmt=WPA-PSK
}
```
This configuration did work on a pi zero (not w) with an external wifi adapter and `drivers_to_load=8192cu`, but not on the zero w.

TwinUsers, 2017-09-17:
I've got it installed on a Pi Zero just fine, didn't need to load any drivers just change network interface in installer-config.txt to wlan0 and added my wpa_supplicant.conf and it booted just fine. I'm using the latest BETA release though :+1:

BenjaminHae, 2017-09-17:
I've just tried it again. `v1.1.1-beta3`, `installer-config.txt`:
```
hostname=artnet_node
packages=unionfs-fuse
ifname=wlan0
timezone=Europe/Berlin
```
In post-install just a `cp` command for setting up my authorized ssh-keys.
And a `wpa_supplicant.conf` file as described above.
Still the same error:
```
=================================================
raspbian-ua-netinst
=================================================
Revision git~caace15
Built on Thu Aug  3 17:47:09 CEST 2017
Running on Raspberry Pi version unknown
=================================================
https://github.com/debian-pi/raspbian-ua-netinst/
=================================================
Starting HWRNG succeeded!
Copying boot files... OK
=================================================
=== Start executing installer-config.txt. ===
=== Finished executing installer-config.txt. ===
=================================================

Network configuration:
  ip4_addr = dhcp
  ip6_addr = disable
  online_config = 

Loading drivers.
Finished loading drivers.
Waiting for wlan0... 
1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. FAILED

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```
It works using raspbian lite.
Maybe there are different models of the `zero w` around?

Thank you all for your time!

TwinUsers, 2017-09-17:
Looks like it's loading drivers fine, just an issue with connecting to the Wi-Fi? 

What did you use to edit the wpa_supplicant.conf? And I'm assuming you've put the .conf file in right folder?

BenjaminHae, 2017-09-17:
I think it was `vim` originally. As the device is not even in the device list I think the issue is not with the wifi connection. My configuration works perfectly well with a pi zero (not w) and an external wifi adapter.

BenjaminHae, 2017-10-01:
I finally found out I can activate wifi by installing the package `firmware-brcm80211`, but obviously only after installation.

Mausy5043, 2017-10-01:
Well, it is in the installer: https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/update.sh#L26

BenjaminHae, 2017-10-01:
`git~caace15` is this version isn't it? So I'm pretty sure. This is really weird. I'll look into it when I have time again.

Mausy5043, 2017-10-01:
Yes that's the `beta3`.

Remarkable 😕

Zeerix, 2017-10-03:
> Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...

Press ENTER and type `dmesg`, the output might be helpful.

Mausy5043, 2017-12-09:
Closing this issue for now, assuming it is resolved. If you feel the closure is in error, please feel free to re-open and add new information.

closed 2017-12-09 by Mausy5043