debian-pi/raspbian-ua-netinst #495: `lftp` returns error : "Cannot assign requested address"

Mausy5043, 2017-10-07:
Trying to use `lftp` to upload data to my webserver is unsuccesful:

```
$ lftp -d mysite.nl
lftp ftpuser@mysite.nl@mysite.nl:~> ls
---- Connecting to mysite.nl (xxxx:xxxx:x::xx) port 21
**** connect(control_sock): Cannot assign requested address
---- Closing control socket
ls: Cannot assign requested address
lftp ftpuser@mysite.nl@mysite.nl:~> ls
---- Connecting to mysite.nl (xxxx:xxxx:x::xx) port 21
**** connect(control_sock): Cannot assign requested address
---- Closing control socket
ls: Cannot assign requested address
lftp ftpuser@mysite.nl@mysite.nl:~> exit
```

Using the same settings and commands on another Debian machine (Ubuntu 16.04) **is** succesfull.  
I expect a package is missing but haven't been able to figure out which one.  
The problem has persisted for the past couple of months.  
Using a build of the current (07OCT2017) `v1.1.x` branch still doesn't fix the problem.

kpfleming, 2017-10-07:
Is ```lftp``` IPv6-aware? It may not be, and it might be trying to use an IPv4 socket with an IPv6 address (the address you masked out looks like an IPv6 format).

Mausy5043, 2017-10-08:
Yes it's an IPv6 address.

brunetton, 2018-03-06:
Same issue here, with a demain name that resolves as a IPV4 adress. Working well on another Debian computer. 
Solved using IP adress instead of domain name.

Mausy5043, 2018-03-08:
Finally seem to have solved it.

Adding this line to `~/.lftprc` : 
```
set dns:order "inet inet6"
```
seems to work.

If someone could confirm this. I can close the issue.

goranche, 2018-03-08:
@Mausy5043 this isn't really related to the installer

closing for now

closed 2018-03-08 by goranche

brunetton, 2018-03-12:
Thanks **SOOOOOOO** much @Mausy5043, it works like a charm ! I was stucked for such a long time with this one !
:champagne: