debian-pi/raspbian-ua-netinst #497: PI3 UART/Serial not working

Ahmed-90, 2017-10-12:
Hello,

Just like the title states. i need to connect a GSM (SIM808) to my PI via serial, but no luck..

**I tried:**

- Using raspi-config.
- Manually editing files `/etc/inittab`, `/boot/cmdline and config.txt`.
- Using scripts like [rpi-serial-console](/lurch/rpi-serial-console) and [uart_control](/itemir/rpi_boat_utils/tree/master/uart_control)
- `systemctl disable serial-getty...` etc..

and all the combos from the above

Tried on release beta2&3 and stable no luck

Thanks

Mausy5043, 2017-10-12:
Thank you for your interest in `raspian-ua-netinst`. In order for us to help you, it's useful if you provide as much detail as possible. Please provide the steps you took and the results:

1. What did you expect to happen?
1. What actually happened?
1. What changes did you make to what files? 
1. What was the result?
1. Please provide any logs or screen output that may be relevant.

Mausy5043, 2017-10-12:
Could you try installing `sudo apt install wiringpi` 

What is the output of  `gpio readall`

Ahmed-90, 2017-10-12:
```log
root@pi:~# gpio readall
wiringPiSetup: mmap (GPIO) failed: Operation not permitted 
```
 
```log
root@pi:~# ls -l /dev | grep gpio
crw------- 1 root root    254,   0 Oct 12 03:50 gpiochip0
crw------- 1 root root    254,   1 Oct 12 03:50 gpiochip1
crw------- 1 root root    254,   2 Oct 12 03:50 gpiochip2
crw------- 1 root root    236,   0 Oct 12 03:50 gpiomem 
```

as for your comment.. i expected serial to work.. i tried GSM, native Serial etc.. not working unable to send or receive a single char, no logs/output, simply blank

as for what files/tries everything here: https://elinux.org/RPi_Serial_Connection#Connection_to_a_microcontroller_or_other_peripheral
and here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=141195

Mausy5043, 2017-10-12:
On my Pi3 I get:

```
pi3 ~ $ gpio readall
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
```

Ahmed-90, 2017-10-12:
any idea why ? its just fresh install.. :/

Mausy5043, 2017-10-12:
The links you supplied use the Raspberry Pi Foundation distribution. `raspbian-ua-netinst` uses the Raspbian kernel and Debian packages. Possibly there are critical differences between the two.

Have you tried using the RPF image?

Ahmed-90, 2017-10-12:
I need `raspbian-ua-netinst` for the net config and the config files, it makes it easier for the non tech savvy to install my software 

any guide or instructions for Serial with `raspbian-ua-netinst` ? or is it on by default?

Mausy5043, 2017-10-12:
As far as I'm aware the serial port is working, because some people use the serial terminal in the absence of a monitor.

I'd suggest to try with the RPF image first, since it is known to work for sure. That way you can make sure the hardware set-up works.

Ahmed-90, 2017-10-13:
I finally figured it out.. it turns out the gpio needs a kernel option/flag that's not included in fresh `raspbian-ua-netinst` by default

---

**To just fix the `wiringpi`/`gpio`**

Add `iomem=relaxed` to `/boot/cmdline.txt`

---
### To enable hardware serial

**Change `/boot/cmdline` to (remove the serial console)**

``` 
dwc_otg.lpm_enable=0 console=tty1 elevator=deadline net.ifnames=0 root=/dev/mmcblk0p2 iomem=relaxed rootfstype=ext4 rootwait
```

**Append the following at the bottom of `/boot/config.txt`**

```
dtoverlay=pi3-miniuart-bt
enable_uart=1
```

**Edit `/etc/inittab`**

```
/etc/inittab
#Spawn a getty on Raspberry Pi serial line
#T0:23:respawn:/sbin/getty -L ttyAMA0 115200 vt100 <-- comment this line
```

Mausy5043, 2017-10-13:
Thanks for providing this solution.
Can the issue be closed?

Ahmed-90, 2017-10-13:
Sure :)

closed 2017-10-13 by Ahmed-90