debian-pi/raspbian-ua-netinst #498: udhcpc doesn't succeed on raspberry 3

toy, 2017-10-12:
I built an image using docker from v1.1.x (2cac85cfcaeeeae7d165910c4006359ca6897a88), wrote it on sd card and started the raspberry without adding any custom config. The output gets stuck on:
```
Loading drivers.
Finished loading drivers.
Waiting for eth0...
OK
Configuring eth0 fro IPv4 using DHCP...
```
Removing ` &>/dev/null` after `udhcpc -i $ifname` also outputs `udhcpc (v1.22.1) started` and repeating `Sending discover...`with pauses of several seconds, I give up after few minutes.

When I use the same image on raspberry 2, `udhcpc` succeeds in few seconds and continues installation.

kpfleming, 2017-10-12:
Silly question... have you previously been able to use the network interface on that Pi3, so you know that it's functional?

toy, 2017-10-12:
@kpfleming Not a silly question and you pointed me in the right direction – the MAC address of the raspberry 3 was not properly whitelisted, so the issue is closed. Thanks!

closed 2017-10-12 by toy