debian-pi/raspbian-ua-netinst #499: what sources for apt?

oupala, 2017-10-13:
I compared installed source from an [official raspbian](https://www.raspberrypi.org/downloads/raspbian/) and from a [raspbian unattended netinstaller](https://github.com/debian-pi/raspbian-ua-netinst).

It appears that sources differs:

- official raspbian

```bash
> cat /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi

> cat /etc/apt/sources.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ jessie main ui
```

- raspbian unattended netinstaller

```bash
>cat /etc/apt/sources.list  
deb http://mirrordirector.raspbian.org/raspbian jessie main contrib non-free firmware  

>cat /etc/apt/sources.list.d/raspberrypi.org.list  
deb http://archive.raspberrypi.org/debian jessie main  
```

I then have a few questions:

- Why  does the `rpi` section in the official raspbian has been replaced by the `firmware` section in the raspbian unattended netinstaller?
- Why  does the `ui` section in the official raspbian has been deleted from the raspbian unattended netinstaller? Is it because the raspbian unattended netinstaller is mainly used to install headless servers?
- Why is there two sources ( `mirrordirector.raspbian.org` and `archive.raspberrypi.org`) with the same `main` section for the single raspbian operating system? One should be enough!

kpfleming, 2017-10-13:
The reason there are two repositories is that some packages are produced by the Raspberry Pi Foundation, they are not part of Raspbian.

oupala, 2017-10-13:
Ok, that makes sense.

Is there a place where I can get som more information? Maybe what are the packages made by the Raspberry Pi foundation?

And that does not explain what there is differences between section in the same repo (`firmware` vs `rpi`, and `ui` only one one distro).

Mausy5043, 2017-10-14:
`firmware` is the Raspbian equivalent of RPF's `rpi`

`ui` is needed for desktop GUIs which we don't need.

oupala, 2017-10-15:
@Mausy5043 Do you have any idea why the firmware section is not included on a default raspbian installation? Is it useless? Or dangerous?

I'm asking me this question because I'm currently installing and configuring the `unattended-upgrades` package. And there is sometime some packages that are not upgraded by `unattended-upgrades`.

Currently, it is those 2 packages:

```bash
libraspberrypi-bin/stable 1.20170703-1 armhf [upgradable from: 1.20170427-1~nokernel1]
libraspberrypi0/stable 1.20170703-1 armhf [upgradable from: 1.20170427-1~nokernel1]
```

Should I included the section for this packages in automatic upgrades?

And by the way, I had already asked this question in the RaspberryPi Foundation's forums: [what sources for apt?](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=195341].

Mausy5043, 2017-10-15:
Your `/etc/apt/sources.list` should contain:

```
deb http://mirrordirector.raspbian.org/raspbian jessie main contrib non-free firmware 
```

Mausy5043, 2017-10-15:
The RPF uses their own kernel and therefore doesn't need the packages in `firmware` from raspbian.org.

We use the raspbian.org kernel and need the packages in `firmware`.

oupala, 2017-10-15:
Ok, thanks for your answers.

Here are my configured sources:

```bash
>apt-cache policy | grep release
     release a=now
     release o=Raspberry Pi Foundation,a=stable,n=jessie,l=Raspberry Pi Foundation,c=main
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=firmware
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=main
     release o=Raspbian,a=oldstable,n=jessie,l=Raspbian,c=firmware
     release o=Raspbian,a=oldstable,n=jessie,l=Raspbian,c=non-free
     release o=Raspbian,a=oldstable,n=jessie,l=Raspbian,c=contrib
     release o=Raspbian,a=oldstable,n=jessie,l=Raspbian,c=main
```

And here is my `unattended-upgrades` configuration file:

```
Unattended-Upgrade::Origins-Pattern {
    "o=Raspbian,n=jessie";
};
```

I think that `firmware`, `non-free`, `contrib` and `main` should be included in `unattended-upgrades` upgrades.

oupala, 2017-10-16:
Some details about upgradeable packages:

- for libraspberrypi0:

```bash
>apt-cache policy libraspberrypi0
libraspberrypi0:
  Installé : 1.20170427-1~nokernel1
  Candidat : 1.20170703-1
  Épinglage de paquet : 1.20170427-1~nokernel1
 Table de version :
     1.20170703-1 800
        900 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
 *** 1.20170427-1~nokernel1 800
        900 http://mirrordirector.raspbian.org/raspbian/ jessie/firmware armhf Packages
        -10 http://mirrordirector.raspbian.org/raspbian/ stretch/firmware armhf Packages
        100 /var/lib/dpkg/status
```

- for libraspberrypi-bin:

```bash
>apt-cache policy libraspberrypi-bin
libraspberrypi-bin:
  Installé : 1.20170427-1~nokernel1
  Candidat : 1.20170703-1
  Épinglage de paquet : 1.20170427-1~nokernel1
 Table de version :
     1.20170703-1 800
        900 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
 *** 1.20170427-1~nokernel1 800
        900 http://mirrordirector.raspbian.org/raspbian/ jessie/firmware armhf Packages
        -10 http://mirrordirector.raspbian.org/raspbian/ stretch/firmware armhf Packages
        100 /var/lib/dpkg/status
```

I don't understand why both packages from `archive.raspberrypi.org` are not installed by **unattended-upgrades**.

Mausy5043, 2017-10-16:
Ref this commit: https://github.com/debian-pi/raspbian-ua-netinst/commit/0a0dfc468d9badec366c516f06095ebc8e5450e7

Note also the Pin-Priority

Mausy5043, 2017-12-09:
Assuming question is answered. Closing

closed 2017-12-09 by Mausy5043