debian-pi/raspbian-ua-netinst #500: Local instillation/preset rootfs?

Ahmed-90, 2017-10-15:
I need to setup multiple devices in waves.. my issue is internet being too slow, like 1 device install with only few extra packages like `vim, python, pip, htop, wiringpi`, would take over an hour to finish, and I'm planing on adding a lot more

I don't want to lose the flexibility of `raspbian-ua-netinst` and make hardcopies in the form of SD card image, because i have lots of custom scripts and configs that updates often..

Is there a way to create a base setup using something like `chroot` and `rootfs` that i can include in the SD card OR setup on a local network server?

Mausy5043, 2017-10-15:
Ref:
Issue #443 
Issue #314 
Issue [Foodeas/#63](https://github.com/FooDeas/raspberrypi-ua-netinst/issues/63)

olmari, 2017-10-15:
Also, local mirror would also be an option so bulk of the data stays inside the LAN then... One tutorial here: https://www.howtoforge.com/local_debian_ubuntu_mirror but google is friend also.

Ahmed-90, 2017-10-15:
@olmari 
Yeah i know but 250gb for raspbian mirror is too much for my use case 😉 

---

@Mausy5043 
I was thinking of setting up a web cache proxy, didn't know that APT specific proxies exist.. i think that will be the most efficient solution for now..

I will close this issue and reopen if needed Thanks for the hints guys 😃

closed 2017-10-15 by Ahmed-90