debian-pi/raspbian-ua-netinst #501: Raspbian repo changed kernel version to 4.9.0-4

jorritnutma, 2017-10-17:
Hi, today I've build the zip but first had to change the kernel version from 4.9.0-3 to 4.9.0-4 in update.sh and build.sh because the linux-imagekernel-image-4.9.0-3-rpi is not in the repos anymore. Could you please update the files?

Mausy5043, 2017-10-17:
I'd love it if someone could provide a solution where we don't need to hardcode the kernel package name.

olmari, 2017-10-20:
I wonder would some repo have symlink to latest images, so we could refer to that... I tried to look around  a little but yet to see any obvous method to achieve this...

olmari, 2017-10-21:
For now, I made plain old manual update on pull request https://github.com/debian-pi/raspbian-ua-netinst/pull/502

Mausy5043, 2017-12-18:
Closing issue since the problem is resolved.

JedMeister, 2018-01-03:
> Closing issue since the problem is resolved.

Do you mean the specific issue (wrong kernel version) or the generic issue (hard coded kernel version required)?

FWIW, if Raspbian don't already, the way to fix it properly would be for a meta package to be produced which always depends on the latest kernel. When a new kernel package is released, a new metapackage would be released depending on the new kernel. Obviously that's outside the scope of this project though...

olmari, 2018-01-03:
@JedMeister 
Specific issue.

If only there would be symlink or something similar on the site kernel is downloaded to allways latest kernel, would immediately solve much of this problem.

JedMeister, 2018-01-03:
Cheers. 

Like I say, a generically named Raspbian meta package would be the ideal way to resolve this IMO. It's the way that Debian have always done it AFAIK.

olmari, 2018-01-03:
Metapackage in raspbian itself won't help us here, we need an way to get latest kernel from download server directly... Unless someone else knows better way and programs it :)

Mausy5043, 2018-01-03:
`curl http://archive.raspbian.org/raspbian/pool/main/l/linux-4.9/ |grep linux-image |awk -F\" '{print $2}'` gives: 
```
linux-image-4.9.0-4-rpi-dbg_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi-dbg_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi-dbg_4.9.51-1%2brpi3_armhf.deb
linux-image-4.9.0-4-rpi2-dbg_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi2-dbg_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi2-dbg_4.9.51-1%2brpi3_armhf.deb
linux-image-4.9.0-4-rpi2_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi2_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi2_4.9.51-1%2brpi3_armhf.deb
linux-image-4.9.0-4-rpi_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi_4.9.51-1%2brpi3_armhf.deb
```

`wget -O - http://archive.raspbian.org/raspbian/pool/main/l/linux-4.9/ |grep linux-image |awk -F\" '{print $2}'` gives the same. 

So, at buildtime something should be possible.

Then, `busybox wget -O - http://archive.raspbian.org/raspbian/pool/main/l/linux-4.9/ |busybox grep linux-image |busybox awk -F\" '{print $2}'` gives this:

```
linux-image-4.9.0-4-rpi-dbg_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi-dbg_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi-dbg_4.9.51-1%2brpi3_armhf.deb
linux-image-4.9.0-4-rpi2-dbg_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi2-dbg_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi2-dbg_4.9.51-1%2brpi3_armhf.deb
linux-image-4.9.0-4-rpi2_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi2_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi2_4.9.51-1%2brpi3_armhf.deb
linux-image-4.9.0-4-rpi_4.9.51-1%2brpi1~jessie_armhf.deb
linux-image-4.9.0-4-rpi_4.9.51-1%2brpi3%2bb1_armhf.deb
linux-image-4.9.0-4-rpi_4.9.51-1%2brpi3_armhf.deb
```
which implies that in the installer something should be possible also... ❓

JedMeister, 2018-01-04:
@olmari 
> Metapackage in raspbian itself won't help us here, we need an way to get latest kernel from download server directly..

Of course! Sorry for the noise...

Mausy5043, 2018-01-09:
I'll see if I can draw @plugwash's attention

Mausy5043, 2018-01-09:
Ref: [requested support](https://bugs.launchpad.net/raspbian/+bug/1742205).

Mausy5043, 2018-01-13:
The Raspbian team isn't very forthcoming.

plugwash, 2018-01-25:
I guess I have to spoonfeed you guys ;) (the .txt extension is just to keep github happy)
[getlatestpackage.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/1665392/getlatestpackage.txt)

goranche, 2018-03-08:
@plugwash oh, come on, in python?!? anyone can do that:  shell, baby, shell 😂

😉  a bit more serious: thanks 👍 
I'll try to find some time for this project again and see what's up and what can be done