debian-pi/raspbian-ua-netinst #503: some packages won't update

oupala, 2017-10-21:
I wanted to update my raspbian using the standard way:

```bash
>sudo apt update
[...]
>sudo apt upgrade
ecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Calcul de la mise à jour... Certains paquets ne peuvent être installés. Ceci peut signifier
que vous avez demandé l'impossible, ou bien, si vous utilisez
la distribution unstable, que certains paquets n'ont pas encore
été créés ou ne sont pas sortis d'Incoming.
L'information suivante devrait vous aider à résoudre la situation : 

Les paquets suivants contiennent des dépendances non satisfaites :
 raspberrypi-bootloader-nokernel : Est en conflit avec: raspberrypi-bootloader mais 1.20170703-1 devra être installé
                                   Est en conflit avec: raspberrypi-kernel mais 1.20170703-1 devra être installé
E: Paquets défectueux
>apt list --upgradable
En train de lister... Fait
libraspberrypi-bin/stable 1.20170703-1 armhf [upgradable from: 1.20170427-1~nokernel1]
libraspberrypi0/stable 1.20170703-1 armhf [upgradable from: 1.20170427-1~nokernel1]
>apt-cache policy libraspberrypi-bin
libraspberrypi-bin:
  Installé : 1.20170427-1~nokernel1
  Candidat : 1.20170703-1
  Épinglage de paquet : 1.20170427-1~nokernel1
 Table de version :
     1.20170703-1 800
        900 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
 *** 1.20170427-1~nokernel1 800
        900 http://mirrordirector.raspbian.org/raspbian/ jessie/firmware armhf Packages
        -10 http://mirrordirector.raspbian.org/raspbian/ stretch/firmware armhf Packages
        100 /var/lib/dpkg/status
ops[23:58] ~>apt-cache policy libraspberrypi0
libraspberrypi0:
  Installé : 1.20170427-1~nokernel1
  Candidat : 1.20170703-1
  Épinglage de paquet : 1.20170427-1~nokernel1
 Table de version :
     1.20170703-1 800
        900 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
 *** 1.20170427-1~nokernel1 800
        900 http://mirrordirector.raspbian.org/raspbian/ jessie/firmware armhf Packages
        -10 http://mirrordirector.raspbian.org/raspbian/ stretch/firmware armhf Packages
        100 /var/lib/dpkg/status
```

It means that some packages are broken or uncompatible.

Fo you have the same behaviour?

Note: it is a fresh install.

Mausy5043, 2017-10-22:
>  Note: it is a fresh install.

It is a fresh install of what?
Which release or branch did you use? 
If at all possible please share the contents of your `installer-config.txt` and post-install.txt`.

oupala, 2017-10-22:
It is a default install of raspbian-ua-netinst v1.1.1-beta3. I just copy the installer to the sdcard and reboot.

So, as far as I understand, there is no `installer-config.txt` and no `post-install.txt`

Mausy5043, 2017-10-22:
The output you are getting from `apt-cache policy` looks odd to me. AFAIK there should be no mention of `stretch`.

Please share the contents of your `/etc/apt/sources.list`  and  `/etc/apt/sources.list.d/raspberrypi.org.list`

Mausy5043, 2017-10-22:
Could you please also share the contents of `/boot/raspbian-ua-netinst-*.log`?
Feel free to remove any personal info that might be in there.

oupala, 2017-10-22:
You're right, I've added stretch repo after the install:

```bash
>more /etc/apt/sources.list.d/mirrordirector_raspbian_org_raspbian.list 
deb http://mirrordirector.raspbian.org/raspbian stretch main firmware
```

But as you can see above, its priority of -10 makes it not eligible for update.

I just wonder why `libraspberrypi-bin` and `libraspberrypi0` cannot be upgraded.

And about your last message, I have no `/boot/raspbian-ua-netinst-*.log` file in `/boot/`.

Mausy5043, 2017-10-22:
You can't use both `jessie` AND `stretch` sources on a single system. That s going to end in disaster for sure.

Please remove the `stretch` source.
Then retry `sudo apt update` and `sudo apt-upgrade` (I'm assuming no packages have been upgraded yet to `stretch`; if they have you'll need to re-install)

`libraspberrypi-bin` and `libraspberrypi0` are pinned to prevent the system from "upgrading" to the  RPF kernel. The error you got:
```
Les paquets suivants contiennent des dépendances non satisfaites :
 raspberrypi-bootloader-nokernel : Est en conflit avec: raspberrypi-bootloader mais 1.20170703-1 devra être installé
                                   Est en conflit avec: raspberrypi-kernel mais 1.20170703-1 devra être installé
E: Paquets défectueux
```

is `apt` trying to remove `raspberrypi-bootloader-nokernel` and replacing it with the RPF kernel in the aptly named `raspberrypi-kernel` and `raspberrypi-bootloader`.  We don't want that.

Mausy5043, 2017-10-31:
Closing the issue, assuming this is resolved. Please feel free to re-open if you wish to add more information.

closed 2017-10-31 by Mausy5043

oupala, 2017-11-05:
Unfortunately, I **have** to use stretch for one package only. The whole distro sticks with jessie.

I know it can be dangerous and I'm using the stretch release for a python package that do not have strong dependencies.

The information I was missing is:

> libraspberrypi-bin and libraspberrypi0 are pinned to prevent the system from "upgrading" to the RPF kernel.

I checked that I do not have the problem until I don't have stretch in my repos. I'll have to check what is the best way to add stretch to my repos.

Thanks @Mausy5043 for the tips.

oupala, 2017-11-07:
In fact, commit 0a0dfc4 is a problem for me as I'm obliged to use stretch repos in addition to jessie repos.

I need stretch as I need [radicale 1.1](https://packages.debian.org/search?keywords=radicale&searchon=names&suite=all&section=all&sourceid=mozilla-search) and [certbot](https://packages.debian.org/search?suite=all&section=all&arch=any&searchon=names&keywords=certbot) (for letsencrypt). These packages are not available in jessie and work very well on a jessie system.

De you think pinning for following packages could be refined?

```bash
$ more /etc/apt/preferences.d/01vcgencmd.pref 
Package: libraspberrypi-bin libraspberrypi0 libraspberrypi-dev libraspberrypi-doc
Pin: release o=Raspbian,c=firmware
Pin-Priority: 800
```