debian-pi/raspbian-ua-netinst #504: Bootloop after 4 seconds of activity.

positron96, 2017-12-24:
I have a raspberry 1 B and trying to install netinst version from scratch. I copied the archive to SD card, did not write any `installer-config.txt` and did not change anything. Upon boot, I get the following logs in the serial:
```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-1-rpi (debian-kernel@lists.debian.org) (gcc version 4.9.2 (Raspbian 4.9.2-10) ) #1 Debian 4.4.6-1+rpi14 (2016-05-05)
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B Rev 2
[    0.000000] cma: Reserved 8 MiB at 0x1b000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0xf bcm2708.serial=0x9ecaef83 smsc95xx.macaddr=B8:27:EB:CA:EF:83 bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  consoleblank=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 431244K/458752K available (6326K kernel code, 449K rwdata, 1972K rodata, 396K init, 741K bss, 19316K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0822c4c   (8300 kB)
[    0.000000]       .init : 0xc0823000 - 0xc0886000   ( 396 kB)
[    0.000000]       .data : 0xc0886000 - 0xc08f64b0   ( 450 kB)
[    0.000000]        .bss : 0xc08f64b0 - 0xc09af99c   ( 742 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000072] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000172] bcm2835: system timer (irq = 27)
[    0.000475] Console: colour dummy device 80x30
[    0.001357] console [tty1] enabled
[    0.001409] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060338] pid_max: default: 32768 minimum: 301
[    0.060751] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060824] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061958] Disabling cpuset control group subsystem
[    0.062064] Initializing cgroup subsys io
[    0.062131] Initializing cgroup subsys memory
[    0.062215] Initializing cgroup subsys devices
[    0.062277] Initializing cgroup subsys freezer
[    0.062334] Initializing cgroup subsys net_cls
[    0.062389] Initializing cgroup subsys perf_event
[    0.062447] Initializing cgroup subsys net_prio
[    0.062504] Initializing cgroup subsys pids
[    0.062641] CPU: Testing write buffer coherency: ok
[    0.062758] ftrace: allocating 21745 entries in 64 pages
[    0.179958] Setting up static identity map for 0x81c0 - 0x821c
[    0.181774] devtmpfs: initialized
[    0.189735] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.190522] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.192410] pinctrl core: initialized pinctrl subsystem
[    0.193471] NET: Registered protocol family 16
[    0.198889] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.200202] cpuidle: using governor ladder
[    0.200289] cpuidle: using governor menu
[    0.208003] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.208088] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.208372] Serial: AMBA PL011 UART driver
[    0.208811] 20201000.uart: ttyAMA0 at MMIO 0x20201000 (irq = 81, base_baud = 0) is a PL011 rev2
[    0.588537] console [ttyAMA0] enabled
[    0.593103] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.641890] bcm2835-dma 20007000.dma: DMA legacy API manager at f2007000, dmachans=0x1
[    0.652487] SCSI subsystem initialized
[    0.656649] usbcore: registered new interface driver usbfs
[    0.662392] usbcore: registered new interface driver hub
[... 92 lines cut ...]
[    3.521085] Key type dns_resolver registered
[    3.529710] registered taskstats version 1
[    3.535796] zswap: loaded using pool lzo/zbud
[    3.547219] vc-sm: Videocore shared memory driver
[    3.553838] [vc_sm_connected_init]: start
[    3.561161] [vc_sm_connected_init]: end - returning 0
[    3.568668] hctosys: unable to open rtc device (rtc0)
[    3.575470] of_cfs_init
[    3.580723] of_cfs_init: OK
[    3.587868] Freeing unused kernel memory: 396K (c0823000 - c0886000)
[    3.603471] mmc0: host does not support reading read-only switch, assuming write-enable
[    3.628509] mmc0: new high speed SDHC card at address 1234
[    3.645451] mmcblk0: mmc0:1234 SA04G 3.69 GiB
[    3.658645]  mmcblk0:
[    3.709478] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    3.717831] Indeed it is in host mode hprt0 = 00001101
[    4.000049] usb 1-1: New USB device found, idVendor=0424, idProduct=9512
[    4.008527] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    4.041465] hub 1-1:1.0: USB hub found
[    4.047106] hub 1-1:1.0: 3 ports detected
[   16.283462] reboot: Restarting system
Uncompressing Linux... done, booting the kernel.
``` ... and then again and again.

Any ideas what is happening?

Mausy5043, 2017-12-24:
Is there a *.log file on the SD-card?

positron96, 2017-12-24:
Sorry, I thought that it was a too early stage for a system to log to file and I did not check if there was any file( Then I rolled a minibian image to a SD card and got it working, so now I cannot check log files.

I believe the problem was a very tired SD card. I had to make 2 attempts to roll the image that finally worked (got errors while writing) and the reason I started reinstalling OS was a broken filesystem.

I guess if I stumble upon this again I'll collect some more data and update the issue. It can be closed if there are no more ideas.

Mausy5043, 2017-12-25:
A "defective" SD card was at the top of my list too. 
I've seen many an SD-card that would quit booting on me while it seemed otherwise fully functional when used as a normal storage device. I think that's due to the unavalability of the first sectors on the card. I have not investigated it much. Lots of "ifs" and "buts" lie on that path.

goranche, 2018-03-08:
closing due to inactivity...
if needed, feel free to reopen

closed 2018-03-08 by goranche