debian-pi/raspbian-ua-netinst #506: RPI 3 Not booting

Cryosim, 2018-02-03:
Rainbow logo. and nothing.

Mausy5043, 2018-02-03:
Welcome to GitHub. And welcome to the `raspbian-ua-netinst`.

It would help us help you if you could provide a tad more information.

Please read the [README.md](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/README.md) and the [CONTRIBUTING.md](https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/CONTRIBUTING.md)

CesarPhlebotomo, 2018-02-13:
I am also having this issue. I followed your instructions for Linux installation, using release v1.0.9 and xzcat to explode the archive file onto the SD card. I did not customize the installer. I put the SD card in the Pi 3, plugged in an HDMI monitor, ethernet cable, mouse, keyboard, then power supply. I hoped it would 'just work', in spite of your documentation not listing support for a Pi 3, but saw the Pi hanging on the rainbow screen again (image attached).
[image]

kpfleming, 2018-02-13:
The 1.0.9 version will not work on an RPi3. You will need to use the 1.1 branch.

CesarPhlebotomo, 2018-02-13:
Thank you. I was able to successfully install and boot using v1.1.1-beta3.

closed 2018-02-15 by Mausy5043