debian-pi/raspbian-ua-netinst #507: ssh key generated on install

oupala, 2018-03-06:
When installing raspbian, the installer generates a server ssh key that is of type ECDSA key (hosted on `/etc/ssh/ssh_host_ecdsa_key`).

ECDSA is known to be weak and vulnerable.

Is there a way to disable the following host key on install?

- `/etc/ssh/ssh_host_rsa_key`
- `/etc/ssh/ssh_host_dsa_key`
- `/etc/ssh/ssh_host_ecdsa_key`

This might be related to the official raspbian installer more than on raspbian-ua-netinst. This might also be related to the official debian installer. Please let me know if this is the case.

kpfleming, 2018-03-07:
It's not the installer(s). The `ssh` startup script generates keys when they aren't present, and it can be configured to limit the keys it will generate.

oupala, 2018-03-07:
How can you configure it in an unattended install mode?

Is the installed package `openssh-server`?

goranche, 2018-03-07:
there is no other way than an unattended install... and SSH will be enabled by default (as documented in the README file)
you can configure the installed system in `post-install.txt`, which is a normal shell script (it will be run using busybox's `sh`). again, this is documented in README.md (in the appropriate branch, of course, probably v1.1.x)

oupala, 2018-03-07:
Great, thanks @goranche for your answer.

And if I want to fix the problem upstream, where should I try fot fix it? In `openssh-server`?

BenjaminHae, 2018-03-07:
Could you please provide sources for your claim that

>  ECDSA is known to be weak and vulnerable.

As far as I known there are only issues concerning implementations of this algorithm.

Elliptic curves have one advantage that really has value on a pi in comparison to RSA. They are a lot faster.

If you want to go further with this issue the openssh project is the way to go: https://www.openssh.com/report.html
Could you please provide a link here, if you do?

goranche, 2018-03-07:
@oupala well, if you really feel there is something to fix, yes, openssh would be the place to go...

but I don't think there is anything to fix, really 😉

I'm closing this issue for now... if you think there is more to be said, comment below, and there is always the option of reopening the issue 👍

closed 2018-03-07 by goranche

oupala, 2018-03-07:
@BenjaminHae Here are the sources I used:

- https://www.guillaume-leduc.fr/securiser-secure-shell-ssh.html
  - http://security.stackexchange.com/questions/5096/rsa-vs-dsa-for-ssh-authentication-keys/46781#46781
  - https://events.ccc.de/congress/2010/Fahrplan/attachments/1780%5F27c3%5Fconsole%5Fhacking%5F2010.pdf
- https://confs.imirhil.fr/20170513_root66_securite-admin-sys/#24

Sorry, these sources are in french!

It is said that ECDSA is linked to the NIST and hence should be disabled.

goranche, 2018-03-08:
This discussion is not related to the installer. Please take it to the appropriate location, as said before, the openssh project would be the right place to go.

oupala, 2018-03-08:
@goranche I agree with you. I was just replying to the question of @BenjaminHae.