debian-pi/raspbian-ua-netinst #508: Model 3B+ support

kpfleming, 2018-03-15:
I've just ordered a pair of 3B+ Pis. When they arrive I'll try an installer built from the master branch, and use this issue to record any problems that occur (and link to PRs to fix them). Hopefully it will only require updating model and CPU information.

kpfleming, 2018-03-26:
This can be closed, at least from my point of view. The required changes include switching to the RPF kernel and bootloader packages, and the raspberrypi-ua-netinst fork of this project already does that (as well as supporting Debian stretch and a number of other improvements).

CryptoPietje, 2018-03-30:
> the raspberrypi-ua-netinst fork of this project

which one?

diederikdehaas, 2018-03-30:
> the raspberrypi-ua-netinst fork of this project

Which one?

kpfleming, 2018-03-30:
https://github.com/FooDeas/raspberrypi-ua-netinst

diederikdehaas, 2019-12-11:
Technically it works (now?) with Model 3B+, but far from perfect.
It boots and installs a system, but f.e. detecting the type fails and thus installs things for all versions. It also looks like it doesn't properly 'do' the leds on the board. But it works.

Sad thing is that I've now concluded that it's VERY unlikely that there will be any kernel or firmware update coming from raspbian (evar). Thus model 4 will not work at all.

So a switch to RPF kernel is needed (see #473) or to (full) Debian as Debian's 4.19 kernel and raspi(3)-firmware package should work too, albeit it. Probably not as good as RPF's kernel+firmware, but it would/could be full Debian. That probably does mean RPi 0/1 won't be supported though.
As I am a Debian 'fanboy', that is the more likely option. It's also more in line with what this project previously has done and Debian's raspi(3)-firmware package installs the firmware files into `/boot/firmware/` so no more errors when updating the kernel f.e. and 64bit (arm64) should also work ...

We'll see.

closed 2019-12-11 by diederikdehaas