debian-pi/raspbian-ua-netinst #511: Pre-downloading the OS

NewFangledMan, 2018-05-18:
I'm attempting to run a full unattended install of Raspbian with a custom password and pre-configured Wifi settings. I have tried a few methods and I believe what I need is to figure out how to do one of two methods shown below

1) I want to pre-download Raspbian onto the SD card and plug it into the PI so that it installs the OS unattended with a custom default user password and a script to host an AP network after install so I can connect to it with SSH through the Pi's own network/run post-install scripts. 
 
2) I attempt to use this software the way it is meant to be used and have the OS download onto the Pi. I don't have ethernet and therefore need to figure out how to set the Pi to connect to the users Wifi and download raspbian with a custom password. Fully install the OS and turn on SSH/run post-install scripts. However past issues have shown that this does not seem to be possible to do completely unattended. The Pi needs to be accessed with a keyboard and some settings updated.

I believe I need one of these two methods to work but I cannot find a solution to properly do either of these options unattended. 

Is there a way?

sullit, 2018-08-04:
afaik you can't do this easily (as in onboard the pi, without any other device shields) without bootstrapping over Ethernet. 

Alternatively you could look at a way to bootstrap the WiFi via BLE from something like an Arduino shield and then pass the WiFi credentials via this to connect to the required WiFi network.

diederikdehaas, 2019-12-11:
Option 2 (installing through wifi) should work.

closed 2019-12-11 by diederikdehaas