debian-pi/raspbian-ua-netinst #512: switching to 'strech' release of raspbian

artynet, 2018-05-19:
Hello Folks,

I was wondering if I could switch to the latest release of Raspbian just changing the release entries in the 👍 

```
scripts/etc/init.d/rcS
```

file. Is that simple as I think ? thanks in advance.

         Kind Regards

edlins, 2018-05-26:
+1
Using `raspberrypi-ua-netinst` to get `stretch` but their reinstall process is broken.  :disappointed:

tomaluca95, 2019-03-31:
I'm tring to install strech but it fails

diederikdehaas, 2019-05-13:
No, that won't work as that version of cdebootstrap-static doesn't know about 'stretch'.
I'm working on a 'buster' update and with that you can install stretch and when the package gets rebuild (by Debian/Raspbian) to pick up a bug fix, you can install buster with it too.

diederikdehaas, 2019-12-11:
Buster update has been merged into the v1.1.x branch

closed 2019-12-11 by diederikdehaas