debian-pi/raspbian-ua-netinst #516: Unable to complete first login

paul-da, 2020-01-05:
I keep getting "Permission denied, please try again."

I've copied the files from the zip in the most recent release (1.1.2) onto the sd card, and not used any additional config. After setup completes, I find the ip address on my router, and using terminal on a mac type:
ssh root@192.168.1.77
I accept the ECDSA fingerprint.
I get the prompt:
root@192.168.1.77's password:
I type:
raspbian
and press enter. I get the message: Permission denied, please try again.
I've tried pi@192.168.1.77 and password of raspberry but that gives me the same response.

Apologies if I'm missing something obvious here. Do I need to specify the password in config first?

Kind regards,

Paul.

diederikdehaas, 2020-01-05:
That it fails with 'root' is understandable as there is a bug here: https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L1014

Without an `installer-config.txt` with a setting for `username` (and `userpw`), no normal user will be created: https://github.com/debian-pi/raspbian-ua-netinst/blob/v1.1.x/scripts/etc/init.d/rcS#L1026
So it's logical that that part failed.

diederikdehaas, 2020-01-10:
The issue is fixed in the v1.1.x branch.
I want to make a couple of other changes (deprecation warnings) to it before making a new release (and merging it into master).
But if you build a new installer yourself from the v1.1.x branch, it should now work.

paul-da, 2020-01-11:
Hi, thanks for this. I built the v1.1.x branch and was able to log in with the default root account and no addtional config.
Cheers,
Paul.

closed 2020-01-11 by paul-da

melyux, 2020-11-11:
Can you make a new release with the fix? I just uploaded the latest release without knowing about this issue and was locked out of the Pi.

commit e586a73d (2020-01-10): Fix sed statement for the changes in sshd_config over the years.

Now 'root' can actually login using a password. Really fixes #516.