debian-pi/raspbian-ua-netinst #517: wget: server returned error: HTTP/1.1 404 Not Found

andowinger, 2020-04-05:
I am not so familiar with git and I hope this is the right place for this issue.
I have a raspberry pie 1 B and therefore the release on the masterbranch does not work due to this [issue190](https://github.com/debian-pi/raspbian-ua-netinst/issues/190). 
One suggest fix was this 1.0.7 RC release. But if I execute it, I receive the following error:
wget: server returned error: HTTP/1.1 404 Not Found
(complete log file is attached).
I guess maybe some hardlinked package is not existing anymore?
Any idea how to fix that?
Thanks a lot

[raspbian-ua-netinst-20200405T001039.log](https://github.com/debian-pi/raspbian-ua-netinst/files/4432746/raspbian-ua-netinst-20200405T001039.log)

diederikdehaas, 2020-05-22:
```
Revision git~1bce340
Built on Wed Apr 22 16:51:36 CEST 2015
```
That would be a very old version of the master branch. `git pull` should update that to the current master branch.
Wheezy does seem to be removed from the archives and your old version of the master branch does indeed not know Buster. The current master branch (or release v1.1.2) does know Buster.

diederikdehaas, 2020-12-24:
Is your issue resolved?
Only just now realize that your version was `v1.0.7-RC` and you can't install buster with that version.
There's only one real fix and that is enlarge the `/boot/` partition as it is too small. You can do this with GParted, but depending on your experience, this may or may not be a viable option.
Probably easiest is to backup the files you want to keep and start over with a recent release, where `/boot/` is 128MB by default.

diederikdehaas, 2021-04-09:
No response in 3 months, therefor closing the issue.

closed 2021-04-09 by diederikdehaas