debian-pi/raspbian-ua-netinst #518: Failed to open config file '/bootfs/config/wpa_supplicant.conf'

cmptscpeacock, 2020-05-15:
When booting to use wlan0 on Pi0 (W) I've copied the various files. I get the following:

1.. Starting wpa_supplicant…
Successfully initialized wpa_supplicant
Failed to open config file '/bootfs/config/wpa_supplicant.conf', error: No such file or directory
Failed to read or parse configuration '/bootfs/config/wpa_supplicant.conf'.

Any thoughts?

diederikdehaas, 2020-05-22:
Everything is run as root, so it's very unlikely to be a  permission problem.
The most likely issue is that `wpa_supplicant.conf` isn't present at all or in the wrong location. When you've written the installer to the SD card, the file should be placed in `config/wpa_supplicant.conf`. The `bootfs` directory is where it gets mounted during installation.
A (complete) log file may help.

diederikdehaas, 2020-12-24:
No response in >6 months, thus closing.
If the problem still exists and you do provide more info, I can take another look and reopen this issue.

closed 2020-12-24 by diederikdehaas