debian-pi/raspbian-ua-netinst #520: cp: write error: No space left on device

typhoon71, 2020-06-03:
I went to do a clean install for buster with v 1.1.2, but for some reason I'm stuck with:

"cp: write error: No space left on device"

I tried v 1.1.1 too but no dice.
I formatted the SD with SDFormatter as usual.
The older v 1.0.9 still works fine.

I'm using the very first raspberry pi, one of those made in china, with 256MB of memory.

Can someone help me? Thanks.

istevens, 2020-06-26:
Hi @typhoon71. Were you able to get around this? I pulled out an old Pi to run a basic Minecraft for my kids, and ran into the exact same issue. Glad I'm not the only one.

typhoon71, 2020-06-26:
No, I wasn't able to solve this. 
Going to try again seeing if some boot option changes something, but at this point I'm petty sure we need help from the devs.

typhoon71, 2020-08-18:
Still have the issue, will stay with stretch and use v1.0.9. Sadly the devs don't answer, is this project still active?

diederikdehaas, 2020-09-29:
>The only thing I noticed is that "df -h" tells me tmpfs is just 64k. Could it be that?
I'm using the very first raspberry pi, one of those made in china, with 256MB of memory.

It's very likely a memory issue. I do have a RPi1, but iirc it's still with 1G RAM.
The changes between v1.0.x and v1.1.x are so massive, it'll be impossible to find the cause. And I can't test whether any change would fix this issue.

typhoon71, 2020-10-13:
This (v1.1.x doesn't work with 256MB devices) should be written on the main page, since it's a breaking change.

I find pretty strange finding the cause is "impossible": with some logging it should be possible to find at least where/when it's happening.
Anyway I understand nobody cares about the old 256MB raspberry anymore...

diederikdehaas, 2020-10-14:
If you provide a complete output of `df -h` and `free [-m|-h]`, that could help.
except for `/dev/` (in the installer), tmpfs is normally created as a % of total memory, which could be a problem with low-mem devices.

> with some logging it should be possible to find at least where/when it's happening

It already logs everything to the screen and the logfile. As you already posted that, it's clear it already fails right at the start.
You could move the starting of HWRNG further down in the file and rebuild the installer, but it isn't exactly a memory hog, so I'm not expecting that to be the magical fix.

typhoon71, 2020-10-14:
Will see to that, but it may take a bit since I got no spare monitor right now.

diederikdehaas, 2020-10-15:
If you do a `df -h` + `free [-m|-h]` before and after the HWRNG line that'd help.
Are you getting the _"Oh noes, something went wrong!"_ line from the `fail()` function? If so, outputting those commands there could also provide useful info

diederikdehaas, 2020-12-24:
No response for 2 months, thus closing.
If you actually do provide the requested info, I'm happy to look at it (and reopen this issue). But I'm not going to leave this open for eternity without any response.

diederikdehaas, 2021-07-15:
I'd like to debug this issue, but as I don't have a 256MB device, I'll need people with such a device willing to help.
If you are such a person, please reply. @istevens @typhoon71 

Ideally one can build an installer from source, but I can also provide 'debug' builds.
If you have a serial device/console, then after writing the installer files, open `cmdline.txt`and move `console=tty1` to be directly after `consoleblank=0`, so it'll look like this:
`consoleblank=0 console=tty1 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200`

That should make it easy to provide the full (debug) log output, which I need to find the problem and subsequently find a solution.

typhoon71, 2021-07-15:
Happy to se you're willing to debug this one - got some time now - last year was a mess for me.
I can do the tests on my rpi 256MB, but I'll need some help in how to produce the logs you need.

I usually copy the installer files on the SD, boot the (headless) rpi and wait till it's done.
(I don't have a serial device/console)

Is it enough to change `consoleblank=0 console=tty1 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200` in cmdline.txt?

diederikdehaas, 2021-07-15:
You don't need to modify cmdline.txt; that's only needed when you do have a serial console.

In the default configuration it outputs to the 'normal' console, which you can view if you connect a monitor/TV via the HDMI cable.

typhoon71, 2021-07-15:
mmm, so, what should I do?

diederikdehaas, 2021-07-15:
Get the latest release https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.1.6 and write it to a SDcard and then you can plug that in the RPi and power it up. Assuming the problem still exists, it should halt/poweroff at some point and then you can take the SDcard out and there should be a `raspbian-ua-netinst-<datetimestamp>.log` file on the SDcard. Post/attach that file here.

typhoon71, 2021-07-15:
Tomorrow in the afternoon I'll check  if the issue still exists and let you know.

I can't build easily - I'd rather rely on you to build a debug version
Any common archive format will do - 7zip opens most of them anyway

diederikdehaas, 2021-07-16:
Apparently the max size is 25MB, so I can't provide a 'normal' zip file. But the only thing needed is an updated `installer-rpi-cpio.gz` file, which stays just under that limit. So write the normal installer to the SDcard and then replace the `installer-rpi-cpio.gz`file with the one attached.
But please run it with the normal installer first!

[installer-rpi.cpio.gz mem-debug v1.1.x version 1](https://github.com/debian-pi/raspbian-ua-netinst/files/6825989/installer-rpi.cpio.gz)

For people building from source, here's the patch:
[0001-Add-various-memory-debug-statements-for-issue-520.patch.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/6830052/0001-Add-various-memory-debug-statements-for-issue-520.patch.txt)
Github doesn't support attaching `.patch` files (:roll_eyes:), so I added `.txt` to it.
The base of this patch (and possibly following `.patch` files) is the v1.1.6 tag/release.

diederikdehaas, 2021-07-16:
I've also prepared another patch, which modifies the amount of memory the gpu uses based on the amount of memory of the model. Once again do `s/\.txt//` on the filename.
[0002-Specify-alternative-gpu_mem-options-for-256-512MB-mo.patch.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/6830083/0002-Specify-alternative-gpu_mem-options-for-256-512MB-mo.patch.txt)

typhoon71, 2021-07-16:
Issue still happens with raspbian-ua-netinst-v1.1.6.

Log from raspbian-ua-netinst-v1.1.6 : [raspbian-ua-netinst-19700101T000009.log](https://github.com/debian-pi/raspbian-ua-netinst/files/6832623/raspbian-ua-netinst-19700101T000009.log)

Log from raspbian-ua-netinst-v1.1.6, with replaced installer-rpi.cpio: [raspbian-ua-netinst-19700101T000013.log](https://github.com/debian-pi/raspbian-ua-netinst/files/6832624/raspbian-ua-netinst-19700101T000013.log)

Available for other tests.

typhoon71, 2021-07-16:
I noticed that the 1.0.9 release that t worked had smaller installer-rpi1.cpio.gz/installer-rpi2.cpio.gz, while later releases have larger installer-rpi.cpio.gz - is it possible that this is related to the issue?

diederikdehaas, 2021-07-16:
I find the `No space left on device` error quite weird, as `free -k` reports 124036k RAM free, which is ~124MB.
The `Copying boot files...` step copies the contents of the installer partition (the SDcard) to `/bootfs/` which is just a place in RAM. In my tests, the size of `/bootfs/` is 48.7MB, which should easily fit...

It also show 182948k RAM total, which is far lower then 256M*1024=262144k and that's what the 2nd patch is about.
When you write the installer to the SDcard, you'll also see a `config.txt` with the following contents:
```
[pi0]
kernel=kernel-rpi1_install.img
initramfs installer-rpi.cpio.gz
[pi1]
kernel=kernel-rpi1_install.img
initramfs installer-rpi.cpio.gz
[pi2]
kernel=kernel-rpi2_install.img
initramfs installer-rpi.cpio.gz
[pi3]
kernel=kernel-rpi2_install.img
initramfs installer-rpi.cpio.gz
dtoverlay=pi3-miniuart-bt
[pi4]
kernel=kernel-rpi2_install.img
initramfs installer-rpi.cpio.gz
[all]
```
The second patch adds 2 lines to `config.txt`, below the `[all]` line:
```
gpu_mem_256=16
gpu_mem_512=32
```
While the increased size of the installer itself does play a role, it turns out that the newer firmware files use `gpu_mem=64` as default. What the 2nd patch does, is set that value to `16` on devices with 256MB RAM (i.e. yours) and to `32` on devices with 512MB RAM. Devices with >=1G RAM stay at the (new) default of `64`

If you build from source you can apply the patch and build a new installer.
But there's also a much simpler/faster way: after writing the (previous) installer, open the `config.txt` file and add those 2 lines, save the file and put the SDcard in the RPi and power it up.

In your case, there's also an alternative way (as described [here](https://github.com/debian-pi/raspbian-ua-netinst#reinstalling-or-replacing-an-existing-system)):
Get the SDcard out of the RPi and mount the FAT32 partition on your PC. Then move `config-reinstall.txt` to `config.txt` (which replaces it) and open that file in a text editor and add those 2 lines. Save the file and unmount the SDcard.
Then put the SDcard in the RPi and power it up. It should now have 48MB more RAM to 'play' with.

Just to make sure: you didn't put any extra files on the SDcard, right?

typhoon71, 2021-07-16:
raspbian-ua-netinst-v1.1.6, with replaced installer-rpi.cpio, edited config.txt with `gpu_mem_256=16`
`gpu_mem_512=32` - no other file on the SD other than the release archive content.
It's blinking, so I suppose it's working. I'll let you know how it goes, but I think you nailed it.

typhoon71, 2021-07-16:
Something went wrong... 
Here's the log: [raspbian-ua-netinst-20210716T200651.log](https://github.com/debian-pi/raspbian-ua-netinst/files/6833262/raspbian-ua-netinst-20210716T200651.log)

Can do more tests tomorrow.

diederikdehaas, 2021-07-16:
Thanks for testing and reporting back :+1: 
This is good progress, but it still fails and the (last) error message of `E: Internal error: install` isn't very informative.
I'm inclined to think it's still a memory issue, but a hint in that direction would be useful

I've attached a new [installer-rpi.cpio.gz mem-debug v1.1.x version 2](https://github.com/debian-pi/raspbian-ua-netinst/files/6833483/installer-rpi.cpio.gz)
the primary change is that in the `fail` function (which produces the `Oh noes...` message), the memory information is also printed to the console (and also ends up in the logfile). It also logs the output of `fdisk -l` which should give me some indication wrt the SDcard used.

So do the same as you described [earlier](https://github.com/debian-pi/raspbian-ua-netinst/issues/520#issuecomment-881686083), but now use the updated installer-rpi.cpio.gz file.
This *should* fail in exactly the same way, but now (hopefully) also print the memory info when it fails.
Please share the log file again.

Patch file: [0003-Print-fdisk-l-output-and-memory-info-in-fail-functio.patch.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/6833481/0003-Print-fdisk-l-output-and-memory-info-in-fail-functio.patch.txt)

Then you could do the same procedure yet again, but not only edit `config.txt`, but also **delete** `kernel-rpi2_install.img`.
This means that when it comes to the `Copying boot files...` step, it has 6MB to copy less ... to RAM.
In this particular case *you* don't need that kernel. I don't consider that a long-term viable solution, but it may just be the difference between success and failure and as such provides useful info.

typhoon71, 2021-07-17:
raspbian-ua-netinst-v1.1.6
installer-rpi.cpio.gz mem-debug v1.1.x version 2
edited config.txt with `gpu_mem_256=16` & `gpu_mem_512=32`

Here's the log: [raspbian-ua-netinst-20210717T062919.log](https://github.com/debian-pi/raspbian-ua-netinst/files/6834292/raspbian-ua-netinst-20210717T062919.log)

Second log, removing kernel-rpi2_install too: [raspbian-ua-netinst-20210717T074427.log](https://github.com/debian-pi/raspbian-ua-netinst/files/6834380/raspbian-ua-netinst-20210717T074427.log)

diederikdehaas, 2021-07-17:
The fact that it got further after removing `kernel-rpi2_install` indicates that it's indeed a memory problem.
Unfortunately it doesn't (really) show in the memory usage 'dumps'.

I got one more possible fix: [installer-rpi.cpio.gz mem-debug v1.1.x version 3](https://github.com/debian-pi/raspbian-ua-netinst/files/6834836/installer-rpi.cpio.gz)
With this variant it's no longer needed to remove `kernel-rpi2_install` as it no longer keeps the backup of the installer in ram (in `/bootfs`), but as soon as the backup is restored to the new FAT32 partition and those get mounted, a `bind --mount` to that is created to `/bootfs`, but only for 256MB models (as I don't want to modify v1.1.x too much for all models).
This should free up an additional ~48MB of RAM and hopefully that should be enough.

Patch file: [0004-Implement-workaround-for-low-mem-models.patch.txt](https://github.com/debian-pi/raspbian-ua-netinst/files/6834845/0004-Implement-workaround-for-low-mem-models.patch.txt)
(Also modified patch 0003 to `fdisk -l $bootdev` instead of just `fdisk -l`)

typhoon71, 2021-07-17:
raspbian-ua-netinst-v1.1.6
installer-rpi.cpio.gz mem-debug v1.1.x version 3
edited config.txt with `gpu_mem_256=16 `& `gpu_mem_512=32`

red led turns on, then the green one too -but aftert that nothing happens anymore 
(usually the other led would lit, turn off, then lit again a show some activity)

no logs produced this time.
just replacing installer-rpi.cpio.gz mem-debug v1.1.x with version 2 has the thing start "normally"

diederikdehaas, 2021-07-17:
Hmm, weird. I have no idea why that would be the case and it worked in my test (with a 512MB model).
I'll do another test with a new 'full' installer and if that works, I'll make a 'release' on my personal fork and I'll link that in this issue.

The green led indicates disk activity and should therefor blink a lot.

typhoon71, 2021-07-17:
OK, will try that then.

diederikdehaas, 2021-07-17:
Here it is: https://github.com/diederikdehaas/raspbian-ua-netinst/releases/tag/v1.1.7-beta-debug-mem

typhoon71, 2021-07-17:
Success!
v1.1.7-beta-debug-mem works fine - it installs everything just it used to be. 
I just logged on a new shiny buster rpi, but will set it up later - maybe on the release version.

Just 2 questions:
- I see no log file, is that normal?
- what did change from the tests I ran and this one?

diederikdehaas, 2021-07-17:
Yeah \o/

When it succeeds, the log file is written to `/var/log/raspbian-ua-netinst.log`. Only on failure is it written to the FAT32 partition on the SDcard. So yes, it is normal.
AFAICT, nothing changed. Maybe it was just that *some bit fell over*. It could be that if you'd tried it again, it would have succeeded. Or I did my test 'wrong' or my instruction was incomplete.

As others may want to test it as well, I figured that the simplest way was to make a release which contained everything, instead of having to apply multiple patches, which can introduce problems too.
And if the test was successful, I had to make a release anyway, which *should* have the same result, but this way it's as close to the 'real thing' as possible.

typhoon71, 2021-07-17:
dang, I should've known that logs go in /var/log/... totally slipped my mind while I fixed myself on the FAT32 partition

In case you need it, here's the log of the successfull install: [raspbian-ua-netinst.log](https://github.com/debian-pi/raspbian-ua-netinst/files/6835180/raspbian-ua-netinst.log)

Thanks a lot for fixing it.

diederikdehaas, 2021-07-17:
I just released v1.1.7 which fixes this issue, but without all the debug statements: https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.1.7

> will set it up later - maybe on the release version.

You can do that, but it won't have a benefit to you. It would only verify that I didn't mess up applying the fixes 'officially'.
 

The full fix is not yet part of the `master` or `v1.2.x` branch, only the special `gpu_mem_*` settings are.
A full, but different fix will (or at least should) be part of those branches, but I'm still working on that.
It's actually the discoveries during that work that led me back to this issue ... and the resulting fixes ;-)

As this issue was reported against the v1.1.x branch, I do consider this issue fixed and will therefor close it, but feel free to reopen it if that was in error.

closed 2021-07-17 by diederikdehaas

typhoon71, 2021-07-18:
well, won't reinstall it for the nth time then! XD
will let you know how new brach/version fare (probably).
thanks again :)

diederikdehaas, 2021-07-18:
> thanks again :)

You're welcome :)
Thank you for bringing this issue to my attention.
I never encountered it myself, so didn't see a (potential) problem, but I do see it now.

> while later releases have larger installer-rpi.cpio.gz - is it possible that this is related to the issue?

While it wasn't the major issue, it is a good point.
And it will be an issue going forward, so I created this issue: https://github.com/debian-pi/raspbian-ua-netinst/issues/531
The work I'm putting in v1.2.x already made me review previous design choices, issues like these do too :)

> will let you know how new brach/version fare (probably).

That would be appreciated. I'll ping you when there's a version to test so you'll know it's available.

commit cbdef9a9 (2021-07-17): Implement workaround for low-mem models.

The installer makes a backup (in memory) of itself to /bootfs, but after
restore to the new FAT32 partition doesn't remove that memory copy.
The 256MB model is so memory constrained that it fails, likely due to
an OOM problem. For those models (only), remove the RAM copy and when
the new boot partition gets mounted (to /rootfs/boot), make a
'mount --bind' to /bootfs, so if any code assumes the data/files to be
there, it'll still find them, so it won't break any code.

This fixes #520.