debian-pi/raspbian-ua-netinst #521: activate ssh for raspberry pi 2

TFTomSun, 2020-06-12:
Hello,
first of all thanks for that great project. I used it on a raspberry pi 3 already and worked like a charm. But was my old raspberry pi 2, i have the problem that ssh is not active after the installation. When I try to connect with putty I get a connection error. 
I already tried to put the ssh file in the sd card. Do I need to add it via the file registration? If yes, is there no default setting for activating ssh?

flxbt, 2020-06-21:
Yes, really a very neat project! I am using it for years as starting point for Devuan installations on the Raspberry Pi. 

IIUC, the latest (systemd based) Raspbian aka "Raspberry Pi OS" has the sshd disabled by default, so if you don't change the default init system (by adding an "installer-config.txt"), you will need to enable it manually (1). Placing an empty file named "ssh" under /boot/ on the sdcard will enable the sshd (only) at next boot:

$ touch /path/to/sdcard-mountpoint/config/files/boot/ssh

As you can include custom files in your installation (2), it should be possible to have a one-shot sshd active on first auto-reboot after the installation on a systemd based system, too. (This might be a sensible default for systemd based installations?!)

Doing a sysvinit based installation at the moment (which starts the sshd by default), I have noticed that the default sshd configuration does not permit root login with password anymore, so additionally, you'll have to configure a non-privileged user or provide a root_ssh_pubkey (both in the installer-config.txt) to be able to log in remotely.

(1) https://www.raspberrypi.org/documentation/remote-access/ssh/ 
(2) https://github.com/debian-pi/raspbian-ua-netinst#installer-customization -> "Bring your own files"

diederikdehaas, 2020-08-18:
Putting a 'ssh' file is something only applicable to raspberrypi.org images. This installer should have a working network/ssh server OOTB, without having to do anything special.
If it doesn't work, I'll need more details to try to figure out why.

diederikdehaas, 2020-12-24:
No response in over 4 months, thus closing.

closed 2020-12-24 by diederikdehaas